libplctag/libplctag.NET #190: How to recover from socket error 10054 (Modbus TCP)?

fmartens, 2021-09-16:
Hello all,

I am testing the Modbus functionality using an Advantech WISE-4060/LAN module. I am reading 3 coils at a 1 second interval. The communication drops and I noticed **ErrorBusy** exception messages in the logging of my test application. My initial test application used libplctag but I was unable to get more logging than **ErrorBusy**. Then I rewrote the test application to use libplctag.NativeImport and this provided more logging:

```
[current iteration: 731, current handle: 0]: 0
2021-09-16 11:48:37.504 thread(1) tag(12) INFO plc_tag_read:1131 Starting.
2021-09-16 11:48:37.504 thread(1) tag(12) DETAIL mb_read_start:1746 Starting.
2021-09-16 11:48:37.504 thread(1) tag(12) DETAIL mb_read_start:1769 Done.
2021-09-16 11:48:37.518 thread(3) tag(12) INFO create_read_request:1205 Starting.
2021-09-16 11:48:37.519 thread(3) tag(12) DETAIL create_read_request:1207 seq_id=2198
2021-09-16 11:48:37.519 thread(3) tag(12) DETAIL create_read_request:1208 registers_per_request = 2000
2021-09-16 11:48:37.520 thread(3) tag(12) DETAIL create_read_request:1209 base_register = 2
2021-09-16 11:48:37.520 thread(3) tag(12) DETAIL create_read_request:1210 register_count = 1
2021-09-16 11:48:37.520 thread(3) tag(12) INFO create_read_request:1217 preparing read request for 1 registers (of 1 total) from base register 2.
2021-09-16 11:48:37.521 thread(3) tag(12) DETAIL create_read_request:1294 Done.
2021-09-16 11:48:37.533 thread(3) tag(0) DETAIL write_packet:978 Full packet written.
2021-09-16 11:48:37.535 thread(3) tag(0) DETAIL write_packet:979  08 96 00 00 00 06 00 01 00 02
2021-09-16 11:48:37.535 thread(3) tag(0) DETAIL write_packet:979  00 01
2021-09-16 11:48:37.548 thread(3) tag(0) DETAIL read_packet:892 data_needed=4, packet_size=4, read_data_len=6
2021-09-16 11:48:37.549 thread(3) tag(0) DETAIL read_packet:892 data_needed=0, packet_size=4, read_data_len=10
2021-09-16 11:48:37.549 thread(3) tag(0) DETAIL read_packet:917 Received full packet.
2021-09-16 11:48:37.549 thread(3) tag(0) DETAIL read_packet:918  08 96 00 00 00 04 00 01 01 00
2021-09-16 11:48:37.549 thread(3) tag(12) INFO check_read_response:1120 Starting.
2021-09-16 11:48:37.549 thread(3) tag(12) INFO check_read_response:1137 Got read response 2198 of length 10 with payload of size 1.
2021-09-16 11:48:37.550 thread(3) tag(12) DETAIL check_read_response:1138 registers_per_request = 2000
2021-09-16 11:48:37.550 thread(3) tag(12) DETAIL check_read_response:1139 register_offset = 0
2021-09-16 11:48:37.550 thread(3) tag(12) DETAIL check_read_response:1140 byte_offset = 0
2021-09-16 11:48:37.550 thread(3) tag(12) DETAIL check_read_response:1141 copy_size = 1
2021-09-16 11:48:37.551 thread(3) tag(12) DETAIL check_read_response:1152 Read is complete.
2021-09-16 11:48:37.551 thread(3) tag(12) DETAIL check_read_response:1191 Done.
2021-09-16 11:48:37.563 thread(1) tag(12) INFO plc_tag_read:1246 elapsed time 58ms
2021-09-16 11:48:37.565 thread(1) tag(12) INFO plc_tag_read:1264 Done
[current iteration: 731, current handle: 1]: 0
2021-09-16 11:48:37.565 thread(1) tag(13) INFO plc_tag_read:1131 Starting.
2021-09-16 11:48:37.565 thread(1) tag(13) DETAIL mb_read_start:1746 Starting.
2021-09-16 11:48:37.565 thread(1) tag(13) DETAIL mb_read_start:1769 Done.
2021-09-16 11:48:37.579 thread(3) tag(13) INFO create_read_request:1205 Starting.
2021-09-16 11:48:37.580 thread(3) tag(13) DETAIL create_read_request:1207 seq_id=2199
2021-09-16 11:48:37.580 thread(3) tag(13) DETAIL create_read_request:1208 registers_per_request = 2000
2021-09-16 11:48:37.581 thread(3) tag(13) DETAIL create_read_request:1209 base_register = 3
2021-09-16 11:48:37.581 thread(3) tag(13) DETAIL create_read_request:1210 register_count = 1
2021-09-16 11:48:37.581 thread(3) tag(13) INFO create_read_request:1217 preparing read request for 1 registers (of 1 total) from base register 3.
2021-09-16 11:48:37.581 thread(3) tag(13) DETAIL create_read_request:1294 Done.
2021-09-16 11:48:37.594 thread(3) tag(0) DETAIL write_packet:978 Full packet written.
2021-09-16 11:48:37.596 thread(3) tag(0) DETAIL write_packet:979  08 97 00 00 00 06 00 01 00 03
2021-09-16 11:48:37.596 thread(3) tag(0) DETAIL write_packet:979  00 01
2021-09-16 11:48:37.609 thread(3) tag(0) DETAIL read_packet:892 data_needed=4, packet_size=4, read_data_len=6
2021-09-16 11:48:37.610 thread(3) tag(0) DETAIL read_packet:892 data_needed=0, packet_size=4, read_data_len=10
2021-09-16 11:48:37.610 thread(3) tag(0) DETAIL read_packet:917 Received full packet.
2021-09-16 11:48:37.610 thread(3) tag(0) DETAIL read_packet:918  08 97 00 00 00 04 00 01 01 00
2021-09-16 11:48:37.610 thread(3) tag(13) INFO check_read_response:1120 Starting.
2021-09-16 11:48:37.611 thread(3) tag(13) INFO check_read_response:1137 Got read response 2199 of length 10 with payload of size 1.
2021-09-16 11:48:37.611 thread(3) tag(13) DETAIL check_read_response:1138 registers_per_request = 2000
2021-09-16 11:48:37.611 thread(3) tag(13) DETAIL check_read_response:1139 register_offset = 0
2021-09-16 11:48:37.611 thread(3) tag(13) DETAIL check_read_response:1140 byte_offset = 0
2021-09-16 11:48:37.612 thread(3) tag(13) DETAIL check_read_response:1141 copy_size = 1
2021-09-16 11:48:37.612 thread(3) tag(13) DETAIL check_read_response:1152 Read is complete.
2021-09-16 11:48:37.612 thread(3) tag(13) DETAIL check_read_response:1191 Done.
2021-09-16 11:48:37.624 thread(1) tag(13) INFO plc_tag_read:1246 elapsed time 59ms
2021-09-16 11:48:37.627 thread(1) tag(13) INFO plc_tag_read:1264 Done
[current iteration: 731, current handle: 2]: 0
2021-09-16 11:48:38.642 thread(1) tag(11) INFO plc_tag_read:1131 Starting.
2021-09-16 11:48:38.642 thread(1) tag(11) DETAIL mb_read_start:1746 Starting.
2021-09-16 11:48:38.643 thread(1) tag(11) DETAIL mb_read_start:1769 Done.
2021-09-16 11:48:38.658 thread(3) tag(11) INFO create_read_request:1205 Starting.
2021-09-16 11:48:38.658 thread(3) tag(11) DETAIL create_read_request:1207 seq_id=2200
2021-09-16 11:48:38.659 thread(3) tag(11) DETAIL create_read_request:1208 registers_per_request = 2000
2021-09-16 11:48:38.659 thread(3) tag(11) DETAIL create_read_request:1209 base_register = 1
2021-09-16 11:48:38.660 thread(3) tag(11) DETAIL create_read_request:1210 register_count = 1
[... 43 lines cut ...]
[current iteration: 732, current handle: 1]: 0
2021-09-16 13:10:46.399 thread(1) tag(13) INFO plc_tag_read:1131 Starting.
2021-09-16 13:10:46.399 thread(1) tag(13) DETAIL mb_read_start:1746 Starting.
2021-09-16 13:10:46.399 thread(1) tag(13) DETAIL mb_read_start:1769 Done.
2021-09-16 13:10:47.163 thread(3) tag(0) WARN socket_read:1229 socket read error rc=-1, errno=10054
2021-09-16 13:10:47.163 thread(3) tag(0) WARN read_packet:909 Error, PLCTAG_ERR_READ, reading socket!
2021-09-16 13:10:47.400 thread(1) tag(13) WARN plc_tag_read:1236 Read operation timed out.
2021-09-16 13:10:47.400 thread(1) tag(13) INFO plc_tag_read:1246 elapsed time 1000ms
2021-09-16 13:10:47.401 thread(1) tag(13) INFO plc_tag_read:1264 Done
2021-09-16 13:10:52.169 thread(3) tag(0) WARN socket_read:1229 socket read error rc=-1, errno=10054
2021-09-16 13:10:52.169 thread(3) tag(0) WARN read_packet:909 Error, PLCTAG_ERR_READ, reading socket!
```

_<<< And so on >>>_

The code I am using is similar to the code provided at: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20NativeImport/NativeImportExample.cs
and my connection string reads: **"protocol=modbus_tcp&gateway=10.0.0.1:502=&path=0&elem_count=1&name=co1&debug=4"**

Is this expected behaviour and should I recover from it by disposing the tags (libplctag) or destroying the tags (libplctag.NativeImport) or is there something else I can do to recover from this.

As a side node, I am using libplctag also with an Allen Bradley ControlLogix and I have not experienced issues like this with the Allen Bradley.

I tested this on versions:
- https://github.com/libplctag/libplctag.NET/releases/tag/libplctag-v1.0.4
- https://github.com/libplctag/libplctag.NET/releases/tag/libplctag-v1.0.8

jkoplo, 2021-09-16:
@kyle-github can you weigh in a bit? I don't think @timyhac or I have ever had the occasion to test Modbus functionality...

kyle-github, 2021-09-17:
I just got back tonight from traveling for work.

Under Windows (I assume this is Windows), that error usually means that the other end of the TCP connection forcibly terminated the connection.   In this case that means that Windows thinks the Modbus device closed the socket.   Here's MSDN on the subject:

**WSAECONNRESET 10054**  Connection reset by peer.  An existing connection was forcibly closed by the remote host. This  normally results if the peer application on the remote host is suddenly  stopped, the host is rebooted, the host or remote network interface is  disabled, or the remote host uses a hard close (see setsockopt  for more information on the SO_LINGER option on the remote socket).  _This error may also result if a connection was broken due to keep-alive  activity detecting a failure while one or more operations are in  progress._ Operations that were in progress fail with WSAENETRESET.  Subsequent operations fail with WSAECONNRESET.

I also highlighted an interesting note about keep-alives.   We should not be running into that problem but... Windows.

Clearly there is at least one bug in the core library: it should understand that the error means that the TCP connection is no longer usable and attempt to reconnect.  That is not happening and that is a problem.   I raised issue [279](https://github.com/libplctag/libplctag/issues/279) on the core library for this.

That leaves us with two parts here:

1. The core library should notice the problem and reconnect to the device.
2. The device is closing the connection for some reason.

I can work on 1 but I cannot help much with 2.  I could see some devices overloading if you were hammering them.  But three coils each second?   That is not a high load.  It does not seem like the device should be resetting the connection.

I did a bit of searching on the Internet and have not found anything obvious yet.   That said, I do see that another library that supports Modbus in Java also seems to show some similar issues under Windows.   

Thanks for bringing this up!

kyle-github, 2021-09-17:
In the mean time, you should be able to recover by destroying all tags pointing to that device and recreating them.   That is not acceptable behavior on the part of the core library, but until I get a fix for this, that is all I have :-(

jkoplo, 2021-09-17:
A couple thoughts:

1. I had an issue with a connection that kept being forcibly closed that ended up being a network-security packet shaping thing on a corporate network. It didn't like my traffic. Are you connected directly to the modbus device? Can you try some modbus reads with a different app/utility?
2. The .NET wrapper probably needs a way to pass through the verbose logging from the native library. I opened #191 to look into this.

kyle-github, 2021-09-18:
I will be releasing version 2.3.7 to get the Windows perf fix (workaround) out.   Once I release that, the prerelease branch will be open for work again and I can give this a try.   

I did find some mentions on the web site for a Java Modbus library that some devices only handle 15 or so requests before closing the connection.   I have not seen that with Moxa Modbus gateways (Modbus TCP to serial).  Either way, the core library should handle this as an error that needs reconnection.

kyle-github, 2021-09-18:
@fmartens I have a possible fix in the fix_279 branch of the core library.   Can you give that a try?  On my tests, I see the tags reconnect.

fmartens, 2021-09-19:
@kyle-github, I will do this on Monday. I need to setup the core project on my system to be able to build it.

fmartens, 2021-09-20:
@kyle-github : I tested it and it seems to recover. Problems start at:
2021-09-20 22:33:07.776 thread(3) tag(0) WARN socket_read:1334 socket read error rc=-1, errno=10054

I have included a partial log below, I can send more if you need it.

```
[current iteration: 127, current handle: 0]: 0
2021-09-20 22:33:06.495 thread(1) tag(12) INFO plc_tag_read:1178 Starting.
2021-09-20 22:33:06.495 thread(1) tag(12) DETAIL mb_read_start:1658 Starting.
2021-09-20 22:33:06.495 thread(1) tag(12) DETAIL mb_read_start:1681 Done.
2021-09-20 22:33:06.497 thread(3) tag(12) INFO create_read_request:1118 Starting.
2021-09-20 22:33:06.497 thread(3) tag(12) DETAIL create_read_request:1120 seq_id=386
2021-09-20 22:33:06.497 thread(3) tag(12) DETAIL create_read_request:1121 registers_per_request = 2000
2021-09-20 22:33:06.497 thread(3) tag(12) DETAIL create_read_request:1122 base_register = 2
2021-09-20 22:33:06.497 thread(3) tag(12) DETAIL create_read_request:1123 register_count = 1
2021-09-20 22:33:06.497 thread(3) tag(12) INFO create_read_request:1130 preparing read request for 1 registers (of 1 total) from base register 2.
2021-09-20 22:33:06.498 thread(3) tag(12) DETAIL create_read_request:1207 Done.
2021-09-20 22:33:06.500 thread(3) tag(0) DETAIL write_packet:881 Full packet written.
2021-09-20 22:33:06.500 thread(3) tag(0) DETAIL write_packet:882  01 82 00 00 00 06 00 01 00 02
2021-09-20 22:33:06.500 thread(3) tag(0) DETAIL write_packet:882  00 01
2021-09-20 22:33:06.502 thread(3) tag(0) DETAIL read_packet:789 data_needed=4, packet_size=4, read_data_len=6
2021-09-20 22:33:06.526 thread(3) tag(0) DETAIL read_packet:789 data_needed=0, packet_size=4, read_data_len=10
2021-09-20 22:33:06.526 thread(3) tag(0) DETAIL read_packet:814 Received full packet.
2021-09-20 22:33:06.526 thread(3) tag(0) DETAIL read_packet:815  01 82 00 00 00 04 00 01 01 00
2021-09-20 22:33:06.527 thread(3) tag(12) INFO check_read_response:1033 Starting.
2021-09-20 22:33:06.527 thread(3) tag(12) INFO check_read_response:1050 Got read response 386 of length 10 with payload of size 1.
2021-09-20 22:33:06.527 thread(3) tag(12) DETAIL check_read_response:1051 registers_per_request = 2000
2021-09-20 22:33:06.527 thread(3) tag(12) DETAIL check_read_response:1052 register_offset = 0
2021-09-20 22:33:06.527 thread(3) tag(12) DETAIL check_read_response:1053 byte_offset = 0
2021-09-20 22:33:06.527 thread(3) tag(12) DETAIL check_read_response:1054 copy_size = 1
2021-09-20 22:33:06.527 thread(3) tag(12) DETAIL check_read_response:1065 Read is complete.
2021-09-20 22:33:06.527 thread(3) tag(12) DETAIL check_read_response:1104 Done.
2021-09-20 22:33:06.528 thread(1) tag(12) INFO plc_tag_read:1293 elapsed time 33ms
2021-09-20 22:33:06.528 thread(1) tag(12) INFO plc_tag_read:1311 Done
[current iteration: 127, current handle: 1]: 0
2021-09-20 22:33:06.528 thread(1) tag(13) INFO plc_tag_read:1178 Starting.
2021-09-20 22:33:06.528 thread(1) tag(13) DETAIL mb_read_start:1658 Starting.
2021-09-20 22:33:06.528 thread(1) tag(13) DETAIL mb_read_start:1681 Done.
2021-09-20 22:33:06.529 thread(3) tag(13) INFO create_read_request:1118 Starting.
2021-09-20 22:33:06.529 thread(3) tag(13) DETAIL create_read_request:1120 seq_id=387
2021-09-20 22:33:06.529 thread(3) tag(13) DETAIL create_read_request:1121 registers_per_request = 2000
2021-09-20 22:33:06.529 thread(3) tag(13) DETAIL create_read_request:1122 base_register = 3
2021-09-20 22:33:06.529 thread(3) tag(13) DETAIL create_read_request:1123 register_count = 1
2021-09-20 22:33:06.529 thread(3) tag(13) INFO create_read_request:1130 preparing read request for 1 registers (of 1 total) from base register 3.
2021-09-20 22:33:06.529 thread(3) tag(13) DETAIL create_read_request:1207 Done.
2021-09-20 22:33:06.531 thread(3) tag(0) DETAIL write_packet:881 Full packet written.
2021-09-20 22:33:06.531 thread(3) tag(0) DETAIL write_packet:882  01 83 00 00 00 06 00 01 00 03
2021-09-20 22:33:06.531 thread(3) tag(0) DETAIL write_packet:882  00 01
2021-09-20 22:33:06.533 thread(3) tag(0) DETAIL read_packet:789 data_needed=4, packet_size=4, read_data_len=6
2021-09-20 22:33:06.533 thread(3) tag(0) DETAIL read_packet:789 data_needed=0, packet_size=4, read_data_len=10
2021-09-20 22:33:06.533 thread(3) tag(0) DETAIL read_packet:814 Received full packet.
2021-09-20 22:33:06.533 thread(3) tag(0) DETAIL read_packet:815  01 83 00 00 00 04 00 01 01 00
2021-09-20 22:33:06.533 thread(3) tag(13) INFO check_read_response:1033 Starting.
2021-09-20 22:33:06.533 thread(3) tag(13) INFO check_read_response:1050 Got read response 387 of length 10 with payload of size 1.
2021-09-20 22:33:06.533 thread(3) tag(13) DETAIL check_read_response:1051 registers_per_request = 2000
2021-09-20 22:33:06.533 thread(3) tag(13) DETAIL check_read_response:1052 register_offset = 0
2021-09-20 22:33:06.533 thread(3) tag(13) DETAIL check_read_response:1053 byte_offset = 0
2021-09-20 22:33:06.533 thread(3) tag(13) DETAIL check_read_response:1054 copy_size = 1
2021-09-20 22:33:06.533 thread(3) tag(13) DETAIL check_read_response:1065 Read is complete.
2021-09-20 22:33:06.533 thread(3) tag(13) DETAIL check_read_response:1104 Done.
2021-09-20 22:33:06.534 thread(1) tag(13) INFO plc_tag_read:1293 elapsed time 6ms
2021-09-20 22:33:06.534 thread(1) tag(13) INFO plc_tag_read:1311 Done
[current iteration: 127, current handle: 2]: 0
2021-09-20 22:33:07.535 thread(1) tag(11) INFO plc_tag_read:1178 Starting.
2021-09-20 22:33:07.535 thread(1) tag(11) DETAIL mb_read_start:1658 Starting.
[... cut ...]
2021-09-20 22:33:46.748 thread(3) tag(11) DETAIL check_read_response:1053 byte_offset = 0
2021-09-20 22:33:46.748 thread(3) tag(11) DETAIL check_read_response:1054 copy_size = 1
2021-09-20 22:33:46.748 thread(3) tag(11) DETAIL check_read_response:1065 Read is complete.
2021-09-20 22:33:46.748 thread(3) tag(11) DETAIL check_read_response:1104 Done.
2021-09-20 22:33:46.749 thread(1) tag(11) INFO plc_tag_read:1293 elapsed time 8ms
2021-09-20 22:33:46.749 thread(1) tag(11) INFO plc_tag_read:1311 Done

```

kyle-github, 2021-09-21:
This is pretty much what I see when I test with my Python Modbus "simulator".   I ^C to kill the server while async_stress.c is running and then find that I have to wait a few seconds before it will reopen the same port.   Once I reopen it, at the next timeout + abort + retry, things go through.   

The log excerpt above (thanks!!!) shows that the library is trying to reconnect several times but there is no open port on the device.   I am still not sure that this is not a problem within Windows or the library that takes a while to time out but I strongly suspect that it is the device that is the cause of the delay.

Will this work for you while I search for better solutions?  I have some other ideas that might help but I will need to do some research and testing.

kyle-github, 2021-09-21:
Well... that was not expected.  I had hoped that doing a shutdown() and then closesocket() (which is supposed to be the "right" way to gently close a socket) would help.  It made the problem worse.  At least with my Python Modbus program it caused it to never be able to reopen a socket on the same port.  This is the fake server/device that was not able to restart after a ^C.   The C library was fine but just kept waiting and retrying and waiting...

Doing a hard closesocket() worked better.

fmartens, 2021-09-21:
I am patient and I can wait. We should not rush the solution I think. 

Regarding the closed port, this is a device issue because I also see connectivity issues on its MQTT and HTTP interfaces. It looks like it does a power-cycle.

kyle-github, 2021-09-22:
My test set up is not equivalent.  I am running the Modbus simulator on the same laptop as the library test program.   So the OS keeps the socket open because the simulator is not responding to the FIN packets that shutdown() sends.

It looks like about all I can do is what I am now doing on this branch: just keep trying.  In your log above, the device is definitely off-line in some manner.

kyle-github, 2021-09-22:
Oops, hit the button too soon. 

I feel confident that the changes I made so far definitely help Modbus.   I am going to go ahead and release this after I release version 2.4.0 or maybe I'll merge these.

fmartens, 2021-09-22:
@kyle-github : I will run my test application again for longer than 24 hours and let you know the outcome but  I need to redirect the logging to file first to be able to filter it a little. I'm OK with merging the changes, all I wanted to say is that I am not in a rush.

fmartens, 2021-10-01:
@kyle-github, I executed a number of tests and the library recovers from the error:
2021-10-01 10:18:10.413 thread(1) tag(0) INFO mb_tag_create:234 Starting.
2021-10-01 10:36:14.664 thread(2) tag(0) WARN modbus_plc_handler:582 Closing socket due to read error PLCTAG_ERR_READ.
2021-10-01 11:04:43.966 thread(2) tag(0) WARN modbus_plc_handler:582 Closing socket due to read error PLCTAG_ERR_READ.
2021-10-01 11:32:58.527 thread(2) tag(0) WARN modbus_plc_handler:582 Closing socket due to read error PLCTAG_ERR_READ.
2021-10-01 12:01:24.140 thread(2) tag(0) WARN modbus_plc_handler:582 Closing socket due to read error PLCTAG_ERR_READ.
2021-10-01 12:29:39.313 thread(2) tag(0) WARN modbus_plc_handler:582 Closing socket due to read error PLCTAG_ERR_READ.
2021-10-01 12:37:47.392 thread(1) tag(0) INFO plc_tag_destroy:1150 Done.

I'm happy with the solution.

fmartens, 2021-10-01:
@timyhac and @jkoplo , I can do more modbus testing if you want to, I still have the device.

timyhac, 2021-10-02:
@fmartens - are you able to work with strings using the existing mapper? My guess is probably not.

kyle-github, 2021-10-02:
I have better Modbus behavior in a new branch.  I am going to merge it this weekend.  I have tested it on Linux and Windows (in a VM) and so far things look good.  I will test it without the existing Windows timer period hack and see if the performance on Windows stays good.

kyle-github, 2021-10-02:
Hmm, performance is better than without the changes but not as good without the timer period hack.   There are still some sleep statements in the common library code.  I need to eliminate those before I can see what the real performance is.

kyle-github, 2021-10-02:
Yeah, those sleep statements are now dominating the time.  Rats.  I was hoping I would be able to tell the difference more easily.

At least the tags are getting serviced very fairly now.

fmartens, 2021-10-03:
> @fmartens - are you able to work with strings using the existing mapper? My guess is probably not.
My devices does not support string but there may be another way to test this.

timyhac, 2021-10-15:
Is there anything else that needs to be resolved on this?

fmartens, 2021-10-15:
@timyhac , it seems you already added the 2.4.0 libplctag libraries to the native export which contains the fix. I will retest the official nuget (and remove my Frankenstein solution) and close this issue.

Thanks for taking care.

fmartens, 2021-10-17:
I updated the NuGet and ran a test. It did not fail and it did not had to reconnect. The NuGet includes the libplctag library that fixes the issue so I'll close the issue.