libplctag/libplctag.NET #191: Enable the native libplctag logging in libplctag.net

jkoplo, 2021-09-17:
This surfaced because of #190 where the developer had to switch to the native wrapper to get the verbose logs out from the libplctag base library. We should make these accessible to developers using libplctag.net without changing their code around.

I haven't looked a ton at pass-through logging, but I know it's available in another open source project I use. I'll probably mirror their implementation if it looks well thought out.

timyhac, 2021-09-17:
Hey @jkoplo - is this the kind of thing you're looking for: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20NativeImport/NativeImportExample.cs#L92

jkoplo, 2021-09-17:
I was actually thinking of the way this library handles exposing logging: https://github.com/chkr1011/MQTTnet/wiki/Trace

Depending on the project, I'm either using NLog or Serilog and I'd probably want libplctag logs flowing into either of those.

Now that Microsoft has the ILogger abstraction, I wonder if any libraries are using that instead of an implementation like in MQTTnet.

timyhac, 2021-09-17:
@jkoplo - yeah nice, a static event on the Libplctag class would be similar to that I think

timyhac, 2021-09-24:
Hi @jkoplo - I had a crack at this and it seems to work OK, see https://github.com/libplctag/libplctag.NET/blob/%23191/src/Examples/CSharp%20DotNetCore/LoggingExample.cs
I think we're on the same page with this?

Doing this I found a bug in NativeImport which has been around for a while and will need to be fixed.

jkoplo, 2021-09-24:
So I actually did a deep dive on the preferred way for libraries (especially nuget packages!) to do logging and it seems like the "correct" way is to make use of `ILogger`. [Masstransit](https://masstransit-project.com/) was held up in some Microsoft GH tickets as a great example of logging in an external library ([Masstransit logging](https://masstransit-project.com/usage/logging.html#basic-configuration)).

I found the skeletons of some old logging abstraction projects that were heavily used and loved before `ILogger` was officially released by MS: [LibLog](https://github.com/damianh/liblog), [common-logging](https://github.com/net-commons/common-logging)

I think we should just adopt `ILogger`. I've got a little more free time lately, I'll try to see if I can prototype it in sometime soon.

timyhac, 2021-10-16:
Heya @jkoplo - I've made a few changes to that branch to make it integrate better with ILogger.

What do you think about doing both: adding the event for each log, as well as integrations with ILogger / Dependency Injection systems. In the branch I've created an extension method that hooks up the LibPlcTag logging as if it were any other log source.
https://github.com/libplctag/libplctag.NET/blob/%23191/src/Examples/CSharp%20DotNetCore/ILoggerExample.cs#L20

I was thinking that we could create nuget packages that provides this functionality for different frameworks - i.e. one for the Microsoft DI system, another for Serilog, another for log4net, not sure what others there are in common use. If we provide separate packages then people can opt in to those integrations.

If they didn't make use of any of these, they could just use the Log event and write it to Console for example: https://github.com/libplctag/libplctag.NET/blob/%23191/src/Examples/CSharp%20DotNetCore/LoggingExample.cs#L13

```csharp
Host.CreateDefaultBuilder(args)
    .ConfigureServices(services =>
    {
        services.AddLibPlcTagLogging();                 // This subscribes to the Libplctag events, and publishes them to an ILogger
        services.AddHostedService<Example>();
    })
    .ConfigureLogging(logging =>
    {
        logging.AddConsole();
        logging.SetMinimumLevel(LogLevel.Information);
    })
    .Build()
    .Run();
```

timyhac, 2021-10-16:
I've also noticed that the logging works if you have access to stderr (like in Console applications for example) by setting the DebugLevel. There was a bug in NativeImport that was fixed in v1.0.18 that prevented this from working.
```csharp
LibPlcTag.DebugLevel = DebugLevel.Spew;
```

timyhac, 2021-10-23:
Hi @jkoplo - I've done another variant of this which exposes an ILogger on the static LibPlcTag class - see Pull Request https://github.com/libplctag/libplctag.NET/pull/206

I'm not sure if this is the best approach, but it works.

Side note: do you know an easy way to get an instance of an ILogger? It seems unless you have Dependency Injection set up, the easiest way is to create a custom implentation. Crazy!?

timyhac, 2021-10-23:
One issue with this is that we need to make sure that the loggers that are injected don't affect the core. İ guess this also would apply to the event delegate solution.

> Logging messages come from deep within the library's core routines. Many of these are very delay sensitive. Do not do anything that would block or delay the return of the logging callback to the library!

İ was thinking we could use a Channel to segregate the consuming and invoking threads.

timyhac, 2021-11-01:
Hi @jkoplo - just wondering if you've had any time to look at this? I'd like to release one of the implementations to production soon.

jkoplo, 2021-11-01:
Give me like a week? My work life has been insane but we just finished a huge technical milestone.

I should have some more time and motivation again.

jkoplo, 2021-11-03:
Spun through it tonight. It seems like you've already got it in `master`?

I see where some of the oddness is - since libplctag is a static class (and manages it's own memory) it really doesn't align with the current paradigm of logging in .NET (context aware, etc). In fact, ILogger (and NLog and Serilog) expect that ALL messages get passed into the logging framework and then the decision about what/where/if to log it is done within the framework. For libplctag, we have to set the logging level we want to filter in code, not in the logging framework.

Tricky stuff. With that said, let's not over design this. I think the way you exposed it with an event delegate is good and fits with our 'thin wrapper' goal. From the delegate, developers can do what they want, including trying to squeeze it into `ILogger` like in your example.

Also, we should increment the minor when we release logging. It's an additional feature and if you depend on logging you can't use previous versions of LibPlcTag.Net without breaking.

timyhac, 2021-11-03:
So @jkoplo v1.1.0?

I agree about the increment minor version - I've also exposed all of the remaining properties onto `Tag`, and some of them onto `Tag<T,M>` which would be included into this version.

The last one I was thinking that could be useful is https://github.com/libplctag/libplctag.NET/issues/211

jkoplo, 2021-11-03:
I'm still getting my head around #211 - gotta get back into the codebase.

If we're going to do a minor anyway, I'd like to get #155 in there - just a set of classes in a separate namespace that hard-code the mapper-tagtype relationship. We've had a couple issues with people getting confused on that...

timyhac, 2021-11-03:
That sounds good @jkoplo. İ looked at that the other day and tried a different solution but i think the dependency injection container solution is better.

İ did have some minor things to add on that but we can discuss if you convert to a PR.

jkoplo, 2021-11-03:
I don't think you can DI it b/c we don't have 1:1...

I think we just need a set of classes like 'DintTag', etc. Basically it's beginner mode. If people need more advanced functionality they'll need to understand the generics and mappers.

timyhac, 2022-02-12:
This one is done

closed 2022-02-12 by timyhac