libplctag/libplctag.NET #213: Uptake core library 2.4.3 to fix memory corruption bug

kyle-github, 2021-11-05:
Hi Tim, Jody,

There was a bad bug lurking in the 2.4.x series.  Thanks to one of the users I was able to determine the root cause (or at least the one I found!) and fix it.   Use after free problem.

I just released 2.4.3 with a fix for the issue.

timyhac, 2021-11-05:
Done

closed 2021-11-05 by timyhac

jkoplo, 2021-11-05:
Less than 90 mins... Not entirely convinced @timyhac isn't a Github Bot...

kyle-github, 2021-11-06:
I keep thinking he doesn't sleep then I remember that we are in very, very different time zones.  But maybe he's just pretending to be in Australia to fool us!

timyhac, 2021-11-06:
Haha @jkoplo my partner thinks so..

But the script makes it pretty straightforward to pull in the binaries from the release page: ./src/copyReleasesIntoNativeImport.ps1