libplctag/libplctag.NET #221: Not able to read or write using the basic example

HenricusL, 2021-11-30:
Hi,

We are trying to use the basic example

        public int ReadDint(string name)
        {
            var tag = new Tag<DintPlcMapper, int>
            {
                Name = “PROGRAM:DataLOG_Test.TestDINT”,
                Gateway = “192.168.20.100”,
                Path = “1,0”,
                PlcType = PlcType.LogixPccc,
                Protocol = Protocol.ab_eip,
                Timeout = 5000
            };
            try
            {
                // Read the value from the PLC
                tag.Read();
            }
            catch (Exception e)
            {
                logger.Error(e);
            }
            return tag.Value;
        }

And get error

libplctag.LibPlcTagException: ErrorUnsupported
   at libplctag.NativeTagWrapper.Initialize()
   at libplctag.NativeTagWrapper.Read()
   at libplctag.Tag`2.Read()

Hints are welcome :)

HenricusL, 2021-11-30:
Hi. This was resolved. Like this it works

public int ReadDint(string name)
    {
        var tag = new Tag<DintPlcMapper, int>
        {
            Name = “TestDINT”,
            Gateway = “192.168.20.100”,
            Path = “1,0”,
            PlcType = PlcType.LogixPccc,
            Protocol = Protocol.ab_eip,
            Timeout = 5000
        };
        try
        {
            // Read the value from the PLC
            tag.Read();
        }
        catch (Exception e)
        {
            logger.Error(e);
        }
        return tag.Value;
    }

closed 2021-11-30 by HenricusL

kyle-github, 2021-12-01:
Actually, this should not work.   The LogixPCCC PLC type is a weird PLC-5 emulation mode on a ControlLogix.   Only PLC-5 data file names can be used (i.e. N7:0 etc.).   I think maybe there is a bug in the .Net library wrapper?

kyle-github, 2021-12-01:
Nevermind, I see that you changed the PLC type later.  My bad.  I need to read the whole thing first!