libplctag/libplctag.NET #222: NativeTagWrapper class is not public. Can't access information from ReadCompleted handler

vernturbine, 2021-12-03:
Hello,

For tag reads I subscribe to the ReadCompleted event and use the ReadAsync() method. This seems to me the cleanest way to read an arbitrary number of tags on separate threads. The problem is that the ReadCompleted event argument "sender" is a NativeTagWrapper object. This is not marked as public and I wasn't able to retrieve information to identify the tag.

It was possible for me to mark NativeTagWrapper and the classes it inherits from as public in the libplctag project. This solved my problem.

Possibly this change could be implemented to the GitHub project?  Also possible is that I am implementing multi-threading in the wrong way, or missing something about ReadCompleted.

Thanks, Vern

timyhac, 2021-12-03:
If you're just looking to read multiple tags concurrently then just await multiple ReadAsync calls:
`await Task.WhenAll(tags.Select(t => t.ReadAsync())`.

If you do specifically want to run multiple threads (I assume you mean Tasks?) then you still don't need to listen for ReadCompleted - just await each ReadAsync call separately.

vernturbine, 2021-12-03:
OK, I figured I was missing something. Thanks for that!

closed 2021-12-03 by vernturbine

timyhac, 2021-12-03:
No worries!