libplctag/libplctag.NET #223: Value cannot be null. (Parameter 'path1')

FROMCITRUS, 2021-12-08:
Using .NET 5 publishing to win-x64
```
Value cannot be null. (Parameter 'path1')
   at System.IO.Path.Combine(String path1, String path2)
   at libplctag.NativeImport.LibraryExtractor.LibraryExists(String folder)
   at libplctag.NativeImport.LibraryExtractor.Init(Boolean forceExtract)
   at libplctag.NativeImport.plctag.ExtractLibraryIfRequired()
   at libplctag.NativeImport.plctag.plc_tag_create(String lpString, Int32 timeout)
   at libplctag.NativeTag.plc_tag_create(String lpString, Int32 timeout)
   at libplctag.NativeTagWrapper.Initialize()
   at libplctag.NativeTagWrapper.Read()
   at libplctag.Tag`2.Read()
```
Which traces to this line.
https://github.com/libplctag/libplctag.NET/blob/ad6abb8f78275d3a6938ca0c4b3033dcef856fae/src/libplctag.NativeImport/LibraryExtractor.cs#L42

This occurs only with a published .exe file. This is caused by https://github.com/libplctag/libplctag.NET/blob/ad6abb8f78275d3a6938ca0c4b3033dcef856fae/src/libplctag.NativeImport/LibraryExtractor.cs#L29

returning an empty string.

My suggestion would be to swap out 
https://github.com/libplctag/libplctag.NET/blob/ad6abb8f78275d3a6938ca0c4b3033dcef856fae/src/libplctag.NativeImport/LibraryExtractor.cs#L31
with
`string location = System.AppContext.BaseDirectory;`
as the AppContext returns a valid location with the publish

timyhac, 2021-12-08:
Interesting, my own application is the same .NET5 published to win-x64 and it works ok. The framework of my application is WPF.

What framework is your application using?

FROMCITRUS, 2021-12-08:
I'm running a WinForms app. My bad, I meant to mention that initially. I am also configured to publish self-contained with ReadyToRun and single file enabled, and I copy over the dll with ForceExtractLibrary set to false.

jkoplo, 2021-12-08:
I've had 'single file' publish play havoc with a lot of path stuff, including my 'appsettings.json' in ASP.NET services.

If the path solution above works that'd be great.

timyhac, 2021-12-16:
There is a pull request that implements this change ready to merge. The only thing left to do is run this change through the environments to see if there are any regressions.

timyhac, 2021-12-31:
Hi @FROMCITRUS - could you try this with latest alpha release?

FROMCITRUS, 2022-01-03:
The libplctag 1.1.0-alpha.1 doesn't seem to have the fix, so I brought in libplctag.NativeImport 1.0.29.
That seems to work as long as it isn't in a `Release` config with `ReadyToRun` compilation enabled. If both of those are set, then an exception occurs on the first usage of NativeImport
```
System.IO.FileNotFoundException: 
File name: 'libplctag.NativeImport, Version=1.0.29.0, Culture=neutral, PublicKeyToken=null'
```
In my specific case, the exception occured on the line `libplctag.NativeImport.plctag.ForceExtractLibrary = false;`
I've been fighting .NET to get any more information out of the exception, but it has been difficult as it only fails in `Release`, not in `Debug`.

jkoplo, 2022-01-03:
I'm actually seeing something similar (though not quite the same) on a single file publish. In debug it gets a path. In a simple release it gets a path. In single file (.net 6) the path returned is null and I get an exception.

I'll check my libplctag native version - I didn't explicitly pull it in so it's probably referencing an older release.

jkoplo, 2022-05-28:
This is related to #259

timyhac, 2023-05-20:
The original issue has been fixed and released, and the second issue seems to have significant overlap with #259 so I will close this one.

closed 2023-05-20 by timyhac