libplctag/libplctag.NET #229: Unable to read string data type tags

childMakarov, 2021-12-18:
I have a Contrologix5000 processor but can't read String data type tags in it. The String tag is in the processor's main tag list.
When I reading an int data type tag, everything is OK, but when I trying to read a String tag I always get the ERROR: _The index and length must refer to positions within the string._
I've tried with .NetFramework4.7.2 and .NetCore3.1, according to the respective code examples of both.  The problem goes the same.
I don't know where goes wrong.
[image]
[image]
.

timyhac, 2021-12-18:
My guess is that [this line of code](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/NativeTagWrapper.cs#L550) is throwing the error.

Are you able to find out exactly what is throwing the error? You might need to clone the repository and use [an example](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ExampleDatatypes.cs#L89). Also would be good to capture the logs and upload here.

U11Leung, 2022-01-08:
We encountered that as well, guess is you are reading non-standard user-defined string type, is that right? According to StringPLCMapper, it accepts 88 bytes lengthed string by definition but you may have a different lengthed one.

System.ArgumentOutOfRangeException
  HResult=0x80131502
  Message=索引和长度必须引用该字符串内的位置。
Arg_ParamName_Name
  Source=mscorlib
  StackTrace:
   在 System.String.Substring(Int32 startIndex, Int32 length)
   在 libplctag.NativeTagWrapper.GetString(Int32 offset)
   在 libplctag.DataTypes.StringPlcMapper.Decode(Tag tag, Int32 offset)
   在 libplctag.DataTypes.PlcMapperBase`1.Decode(Tag tag)
   在 libplctag.Tag`2.DecodeAll()
   在 libplctag.Tag`2.Initialize()
   在 ConsoleApp3.Program.Main(String[] args) 在 C:\Users\......

kyle-github, 2022-01-08:
There may be/probably is a bug in the core C DLL that will allow an overwrite.  I am still trying to figure out how to make sure it is completely fixed and add tests so that it cannot regress.   See this bug [#328.](https://github.com/libplctag/libplctag/issues/328)

kyle-github, 2022-01-08:
The problem will happen when you have non-standard string sizes but are not using the special string attributes to tell the library the correct sizes.  See the wiki section on [optional attributes](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#optional-generic-attributes), particularly the ones starting with `str_`.

U11Leung, 2022-01-09:
Regarding this OutOfRangeException, two facts need further attention, one is that, exception is not always thrown out (even against the same tag in each reading), or say, reading may complete flawlessly, and the readout value is correct; the other is, depsite not fully tested, we believe exception happens only when reading non-standard less-than-82-bytes strings.

kyle-github, 2022-01-09:
I just released a new core C DLL version, 2.4.10, with fixes to prevent heap corruption with non-standard string sizes when the `str_*` tag attributes are not used.

timyhac, 2022-01-10:
This has been released with libplctag.NativeImport v1.0.30

U11Leung, 2022-01-10:
Thanks for your work, how is it going with the other package?

timyhac, 2022-01-10:
Hi @U11Leung - sorry i did not understand your question, could you please rephrase it

U11Leung, 2022-01-10:
I'm sorry, but I asked was https://www.nuget.org/packages/libplctag/ updated too? Seems only NativeImport was updated.

jkoplo, 2022-01-10:
The two packages are set such that libplctag.net takes a dependency on the native package at a version greater than X. Native package updates don't _necessarily_ require the main package to be updated.

To use the latest native package, explicitly add the latest native nuget package to your project.

U11Leung, 2022-01-10:
Well explained and thanks!

closed 2022-01-22 by timyhac