libplctag/libplctag.NET #230: Tag.Write function needs to long

xxxdomi, 2021-12-20:
Hello, I've got an application with hundreds of (input-) tags. Every 10 ms tags can change, which means, that they will be written from different threads to the controller.

I made measurements with one input which was changed and written from different threads to the controller (I used the .net lock to avoid multi-threading problems). The main problem is, that the function Tag.Write() needs from 7 to 11 ms time which blocks my application extremly.

Do you have any idea how I could improve the speed of the write function?

In my example, I am using the following settings:
Path = "1,0", 
PlcType = PlcType.ControlLogix,
Protocol = Protocol.ab_eip,
Timeout = TimeSpan.FromSeconds(5)

Datatype = INT32 - i am using the DintPLCMapper-Class to write the tag.

I am using the latest stable versions of the libraries.

Thanks in adavance,
Domi

timyhac, 2021-12-20:
That seems about right, the performance of the network and the PLC itself is what determines this (not libplctag).
You could think about changing how you're application works, maybe sending multiple values (ie an array) to a queue in the PLC.

kyle-github, 2021-12-21:
+1 @timyhac.

As long as you are using blocking calls, you are going to block your app.  If you call tag.Write() with a zero (0) timeout, then the library will do the work in the background and your thread can go off and do whatever else you need it to.  You will need to check on the status of the write operation before you do another operation on the tag.

The write time of ~10ms sounds pretty good to me.  PLCs are slow.  Note that if you write a lot of data at the same time either via writing more than one tag or writing an array, you can amortize that time.  If your tags are relatively small, you can pack many requests into a single packet to the controller.

closed 2022-01-27 by timyhac