libplctag/libplctag.NET #235: Can we store references to .Net objects and call callbacks with them?

kyle-github, 2022-01-26:
See [PR #345](https://github.com/libplctag/libplctag/pull/345) for more details.

This is a long-awaited feature, but one that I am not sure will work for wrappers for languages that have GC that moves objects.  

The idea is that when you register a callback with the native C DLL, you can pass it a reference to a CLR/.NET object.   When the native DLL calls the callback, one of the arguments it passes back to the CLR will be the CLR object.  This removes a lot of boilerplate code for looking up objects based on tag ID.

However, I do not know what happens with GC.   If the object moves in memory due to GC, then the reference/pointer stored by the native library is going to be invalid and we get a one-way trip to segfault-land.

I have raised a question for Java in the JNA forum, and I am hoping you can figure out what happens with .Net.

timyhac, 2022-01-27:
Yes the [GCHandle](https://docs.microsoft.com/en-us/dotnet/api/system.runtime.interopservices.gchandle?view=netstandard-2.0) class is designed for this scenario.

It should work - the primary libplctag.NET package won't use it because the .NET Tag object wraps the core tag object, and the callback is an instance method so it always has a reference to itself.

If it ends up being merged into core then we can expose it in the NativeImport package - maybe someone else will find a use.

jkoplo, 2022-01-27:
@timyhac has a lot more experience with C interop than I do. I think we might end up using it in the wrapper library at some point. I'm a little disappointed with some of our performance with async/await and I think this _might_ be useful to handle things in a better way.

kyle-github, 2022-02-02:
Version 2.5.0 will be released later this week with the additional API function supporting a void pointer (C type for generic pointer).

It will also have a create event callback but I am not thrilled about how good it is.   It is relatively useful, but does not give you much more than the existing READ completed event.

timyhac, 2022-02-15:
This looks like its done

closed 2022-02-15 by timyhac