libplctag/libplctag.NET #240: Due to the heap corruption fix for the latest Native Import, writing strings no longer works.

redsuinit, 2022-02-09:
Every time you try and write to a string that is not standard length (so a length restricted string) using the Write() functionality you get an ErrorBadConfig.

timyhac, 2022-02-09:
Hi @redsuinit - thanks for the error report, are you able to provide some further detail around how you're triggering this error. Some code would be ideal.

redsuinit, 2022-02-09:
Sure:

                    var myStringTag = new Tag<StringPlcMapper, string>()
                    {
                        Name = "POS680.Stats.TimeStampIn",
                        Gateway = address,
                        Path = path,
                        PlcType = type,
                        Protocol = protocol,
                        Timeout = TimeSpan.FromSeconds(timeout)
                        Value = "String"
                    };

                    // Write the value to the PLC
                    myStringTag.Write();

Now this string is a limited length string, it can only be 20 characters. Previously I could write to this string, but would occasionally get a heap issue. Now it won't do it at all.

kyle-github, 2022-02-09:
Please look through the [wiki page](https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling) on string handling.

There were recently bug fixes in the core library to prevent heap corruption in circumstances.   That is probably what you are hitting.   See issue #233.

redsuinit, 2022-02-09:
Kyle, that does seem to be what I am running into, I guess I just need some guidance on how to handle this through the .NET wrapper. I installed the latest version of the NativeImport and 1.1.0 alpha for the .net wrapper. I'm writing this program in .net core 5.0

timyhac, 2022-02-09:
Yes it looks like I exposed the string configuration on the non generic Tag only.
I still think this makes sense, but this change to core makes the ability to configure the string properties essential. I have a PR to add the ability to allow mappers to make such changes before tag creation which I think is the library solution here.
For @redsuinit ' you'll need to use the nongeneric tag for the time being.

redsuinit, 2022-02-09:
So something like this?

                    var myTag = new Tag()
                    {
                        Name = TagName,
                        Gateway = address,
                        Path = path,
                        PlcType = type,
                        Protocol = protocol,
                        Timeout = TimeSpan.FromSeconds(timeout),
                        StringMaxCapacity = 20,
                        StringTotalLength = 24,
                    };

                    // Write the value from the PLC
                    myTag.SetString(0, "String");
                    myTag.Write();

?

kyle-github, 2022-02-09:
@redsuinit, if there are things you think are missing in the wiki, please let us know!   We need to make sure people understand how strings get weird and how to deal with them!

redsuinit, 2022-02-09:
@kyle-github I'm a Mechanical Engineer by education and trade but I've been squeezed into a developer role so it takes just a little extra grease to get my gears to the right place with coding. As far as the C core wiki, no I think it's a fantastic asset and not much needs to be added. As for the .NET wrapper perhaps an example of the string (as I have above) added to the example code snippits could be useful.

timyhac, 2022-02-10:
Yes agree, all of the examples with strings will need to be updated

AndyMedtronic, 2022-04-25:
I tried this, but the solution code did not work for me. However on investigation I noted that in the generic case tag.Initialize was called, so adding this after the tag construction fixed this

i.e:

 var myTag = new Tag()
                {
                    Name = TagName,
                    Gateway = address,
                    Path = path,
                    PlcType = type,
                    Protocol = protocol,
                    Timeout = TimeSpan.FromSeconds(timeout),
                    StringMaxCapacity = 20,
                    StringTotalLength = 24,
                };

**myTag.Initialize();**

                // Write the value from the PLC
                myTag.SetString(0, "String");
                myTag.Write();

AndyMedtronic, 2022-04-25:
With the non-generic call, the setting of the StringTotal Length gave rise to a BadParam error
So I ended up with a function

`  
        public void WriteNonGenericStringTag(string tagName, string stringValue, int buffLength = 0)
        {

            var stringTag = new Tag
            {
                Name = tagName,
                Gateway = _ipAddress,
                Path = _plcPath,
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(5),
            };

            if (buffLength > 0)
            {
                stringTag.StringMaxCapacity = (uint)buffLength;
             }

            if (!stringTag.IsInitialized)
                stringTag.Initialize();

            stringTag.SetString(0, stringValue);
            stringTag.Write();

        }
`

which did seem to do the job

closed 2022-05-28 by jkoplo