libplctag/libplctag.NET #243: Problem with the .NET Framework example program

BartGuy1215, 2022-03-06:
This is kind of a strange one, and I REALLY need some help figuring this out! I have a VS2019 .NET4 framework based project that needs to now communicate to an AB CLX L81E PLC.

I compiled the .NET Core example and it runs fine on .NET5. Perfect! But my existing project is based on .NET4  Framework (4.8 specifically) and I was having issues with references to the other projects in the solution. Not being an EXPERT VB guy, I saw the example project that used .NET4.7.2 and decided to try it.

Here's the code that is in the example provided (i didnt change anything):

public static void Run()
{
Console.WriteLine($"\r\n*** ExampleRW ***");

        const int TIMEOUT = 5000;

        //DINT Test Read/Write
        var myTag = new Tag<DintPlcMapper, int>()
        {
            //Name of tag on the PLC, Controller-scoped would be just "SomeDINT"
            Name = "MyTESTDint",
            
            //PLC IP Address
            Gateway = "172.16.1.2",

            //CIP path to PLC CPU. "1,0" will be used for most AB PLCs
            Path = "1,0", 

            //Type of PLC
            PlcType = PlcType.ControlLogix,

            //Protocol
            Protocol = Protocol.ab_eip,

            //A global timeout value that is used for Initialize/Read/Write methods
            Timeout = TimeSpan.FromMilliseconds(TIMEOUT), 
        };
        myTag.Initialize();

        //Read tag value - This pulls the value from the PLC into the local Tag value
        Console.WriteLine($"Starting tag read");
        myTag.Read();

        //Read back value from local memory
        int myDint = myTag.Value;
        Console.WriteLine($"Initial Value: {myDint}");

        //Set Tag Value
        myDint++;
        myTag.Value = myDint;

        Console.WriteLine($"Starting tag write ({myDint})");
        myTag.Write();

        //Read tag value - This pulls the value from the PLC into the local Tag value
        Console.WriteLine($"Starting synchronous tag read");
        myTag.Read();

        //Read back value from local memory
        var myDintReadBack = myTag.Value;
        Console.WriteLine($"Final Value: {myDintReadBack}");

The code is exactly the same as the .NET Core example program running on .NET 5, but in the framework example project, I get the following exception when executing the initialize function:

System.IO.FileNotFoundException HResult=0x80070002
Message=Could not load file or assembly 'libplctag.NativeImport, Version=1.0.17.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.
Source=libplctag StackTrace:
at libplctag.NativeTag.plc_tag_create(String lpString, Int32 timeout) in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\NativeTag.cs:line 12
at libplctag.NativeTagWrapper.Initialize() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\NativeTagWrapper.cs:line 184
at libplctag.Tag.Initialize() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\Tag.cs:line 146
at libplctag.Tag`2.Initialize() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\libplctag\TagOfT.cs:line 102
at CSharpDotNetFramework.ExampleRW.Run() in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\Examples\CSharp DotNetFramework\ExampleRW.cs:line 38
at CSharpDotNetFramework.Program.Main(String[] args) in C:\Users\jbart\JDB Team Dropbox\JDB Team Data\Digi\AB PLC AGC\NEW VB HMI\libplctag.NET-master\src\Examples\CSharp DotNetFramework\Program.cs:line 9

The other example projects work fine, but I REALLY to use libplctag on .NET4 Framework.

I'm SURE it's something i'm doing wrong or something on my laptop, I just can't figure out what!! 

Any help is appreciated.

John

timyhac, 2022-03-06:
Try removing the direct reference to the libplctag project and instead use the Nuget package.

BartGuy1215, 2022-03-06:
I’ll google that.  Not sure exactly what Nuget is! 

Thank you.

John

BartGuy1215, 2022-03-06:
Where exactly do I find the NuGet package for libPLCtag??

timyhac, 2022-03-06:
@BartGuy1215 just use [your normal package manager ](https://docs.microsoft.com/en-us/nuget/quickstart/install-and-use-a-package-in-visual-studio)and search for "libplctag"

BartGuy1215, 2022-03-06:
Okay, I THINK I have it.  I will test more thoroughly tomorrow, but for now i'm not getting an exception when initializing the tag.

I will close the issue as I believe it's solved!  Thank you for the help pointing me to NuGet.  And kudos again to Kyle for creating this package.  Let me know how I can contribute to the effort!

John

BartGuy1215, 2022-03-06:
@timyhac Thanks again for the help!  John

closed 2022-03-06 by BartGuy1215