libplctag/libplctag.NET #244: Struggling Reading arrays of Real using libplctag in .NET

BartGuy1215, 2022-03-09:
I thought i was well on my way with a project i'm working on, but I guess not.  

The rest of the project works great!  This library is really nice, but I've hit a wall on a seemingly simple issue.  

Trying to read an array of 1024 Reals from a ControlLogix PLC.  Looking at the example code sent with libplctag in C#, here's where I am:

    Private _PLC_FIFO_TAG As Tag(Of RealPlcMapper, Single())

    Private Sub _CreateFIFOTag()
        ' FIFO tag array is fixed at 256 of type display_fifo_entry = 1024 total
        _PLC_FIFO_TAG = New Tag(Of RealPlcMapper, Single())() With
        {
            .Protocol = libplctag.Protocol.ab_eip,
            .Name = "test_array",
            .Gateway = "172.16.1.2",
            .PlcType = PlcType.ControlLogix,
            .Path = "1,0",
            .Timeout = TimeSpan.FromMilliseconds(5000),
            .ArrayDimensions = New Integer(1024)
        }
        _PLC_FIFO_TAG.Read()
    End Sub

The .ArrayDimensions initializer is not correct and I can't figure out what it's looking for.  Intellisence says it needs an array of integers, but I don't see how that can be the case.  Except for C# code I can't find an example that explains the issue.

Any help would be appreciated!

John

timyhac, 2022-03-09:
Your float array in the PLC is a one dimensional array of length 1024, so the array dimension property should be {1024}.

BartGuy1215, 2022-03-10:
Understood.  Here's where I landed with a piece of test code that I wrote to prove the concept.

Tag in the PLC is array of 6 reals for testing

        _PLC_FIFO_TAG = New Tag(Of RealPlcMapper, Single())() With
        {
            .Protocol = libplctag.Protocol.ab_eip,
            .Name = "TEST_ARRAY",
            .Gateway = "172.16.1.2",
            .PlcType = PlcType.ControlLogix,
            .Path = "1,0",
            .Timeout = TimeSpan.FromMilliseconds(200),
            .ArrayDimensions = New Integer(6 - 1) {}
        }
        _PLC_FIFO_TAG.Initialize()
        _PLC_FIFO_TAG.Value = New Single(6 - 1) {1, 2, 3, 4, 5, 6}
        _PLC_FIFO_TAG.Write()

When stepping through the code .Value has a valid Single array created by the NEW, but after the write is executed exception thrown "errNoData".  I will try to get the code this morning and walk through it.  I use the NeGet package per your previous advice on another issue which dosent install the source, so i can't walk it back completely.

So it looks like as a general statement, for arrays of one, the code provides storage in Value, for multi dimension arrays, we have to push storage into value?  Correct assumption?

John

timyhac, 2022-03-10:
Very often, Tag arrays are 1D arrays, but modern Allen Bradley controllers allow you to define 2D or 3D array tags as well.
`ArrayDimensions` is how you tell libplctag the length of the dimensions of `Value`. i.e. how many dimensions it has, and how long each dimension is.

If, for example, the Tag is not a 1D float array of length 1024, but instead a 3D float array of lengths 8, 8, 8, you would need to configure ArrayDimensions to be 1d integer array of length=3 with value { 8, 8, 8 }.

```vb
_PLC_FIFO_TAG = New Tag(Of RealPlcMapper, Single(,,))() With
{
    .Protocol = libplctag.Protocol.ab_eip,
    .Name = "TEST_ARRAY",
    .Gateway = "172.16.1.2",
    .PlcType = PlcType.ControlLogix,
    .Path = "1,0",
    .Timeout = TimeSpan.FromMilliseconds(200),
    .ArrayDimensions = New Integer() {8, 8, 8}
}
_PLC_FIFO_TAG.Initialize()
_PLC_FIFO_TAG.Value = New Single(8,8,8) {...}
_PLC_FIFO_TAG.Write()
```

BartGuy1215, 2022-03-10:
Hello Tim,

 

Thank you for the feedback, but your example does not compile….

 

 

I’ll keep looking….

BartGuy1215, 2022-03-10:
I think I understand now.  Testing in the next hour or two…

Thanks again for the help.

John

BartGuy1215, 2022-03-10:
Tim,

Thank you again for the help.  Issue is solved.

timyhac, 2022-03-12:
Awesome, thanks!

closed 2022-03-12 by timyhac