libplctag/libplctag.NET #245: AB Micro850 Read is always Timeout

AdamXu23, 2022-03-14:
##  AB Micro850 Read is always Timeout

```csharp
//DINT Test Read/Write
            var myTag = new Tag<DintPlcMapper, int>()
            {
                //Name of tag on the PLC, Controller-scoped would be just "SomeDINT"
                //Name = "PROGRAM:SomeProgram.SomeDINT",
                //Name = "PROGRAM:SomeProgram.SomeDINT",
                Name = "Programs:SomeProgram.SomeDINT",
                //Name = "SomeProgram:SomeDINT",
                //Name = "Programs:GlobalVariables.SomeDINT_1",
                //Name = "SomeDINT_1",

                //PLC IP Address
                Gateway = "192.168.127.2",

                //CIP path to PLC CPU. "1,0" will be used for most AB PLCs
                Path = "1,0", 

                //Type of PLC
                PlcType = PlcType.Micro800,

                //Protocol
                Protocol = Protocol.ab_eip,

                //A global timeout value that is used for Initialize/Read/Write methods
                Timeout = TimeSpan.FromMilliseconds(TIMEOUT)
            };
            myTag.Initialize();
```

[image]
[image]

I can't find the problem

jkoplo, 2022-05-28:
There's apparently some weirdness with Micro850. See if this helps you out: https://github.com/libplctag/libplctag/issues/157

Try leaving off the `Path` property completely.

shuvendu2018, 2022-07-13:
After removing the path from tag variable the problem solved.

closed 2022-09-14 by timyhac