libplctag/libplctag.NET #248: Copyright notice

IsaacSherman, 2022-04-11:
How should we acknowledge use of this library?  There's no mention of copyright, even in the license.

jkoplo, 2022-04-11:
It's MPL licensed, so I don't believe copyright is much of an issue. Obviously I'm not a lawyer, but there should be plenty of general advice on MPL + copyright around.

If you find good guidance, please post back here for future reference. Thanks!

IsaacSherman, 2022-05-09:
Hey, just so you know, I'll be crediting this repo as "Copyright (c) 2022 libplctag.NET contributors".  I'm watching this thread if there's ever a need to change that.

IsaacSherman, 2022-10-11:
The MPL is all well and good, it's a great license.  But even it refers to
not altering the copyright notices.  All I'm looking for is some way to
attribute use of the library; even something as simple as "Copyright (c)
the contributors to the libplctag.NET project" would suffice.  I just want
to get it from the source.  I would humbly suggest adding a line like that
at the top of the license file so everything is in one place.

timyhac, 2022-10-12:
Should be fine to do this- can you elaborate on why you need this?

IsaacSherman, 2022-10-12:
Sure.  My company is releasing free software which includes a simple tag browser using LibPLCTag.  One of the things we're legally required to do is generate a FOSS BOM- a Free and Open Source Software Bill of Materials.  In short, that means listing and crediting the creators of all of the software we've used which we didn't write.  Most of the licenses I've read have had the copyright notice at the top, or prominently displayed somewhere (even on the NuGet package in some cases).

timyhac, 2022-10-12:
I see - MPL v2 doesn't require attribution as I understand, so this legal requirement must be from elsewhere, you do however need to tell recipients where they can get the source code.

Its great that you're focusing on this - so many businesses just don't care about compliance with respect to open source they use.

It looks like your 2nd comment is posted 9th May. Your 3rd comment on this issue was posted to github yesterday.

closed 2022-10-12 by timyhac