libplctag/libplctag.NET #252: NotImplementedException

EffectoSolutions, 2022-04-28:
Hi,

I try to use this library instead an old libPLCTagWrapper that I have in one of our project. I downloaded the nuget and implement the method but I run a NotImplementedException on the NativeTagWrapper.coreLibEventCallback

I tried to find what's wrong in the the source code, and it seems that the Event catched is not in the switch case of the method.

```
void coreLibEventCallback(int eventTagHandle, int eventCode, int statusCode)
        {

            var @event = (Event)eventCode;
            var status = (Status)statusCode;
            var eventArgs = new TagEventArgs() { Status = status };

            switch (@event)
            {
                case Event.ReadCompleted:
                    ReadCompleted?.Invoke(this, eventArgs);
                    break;
                case Event.ReadStarted:
                    ReadStarted?.Invoke(this, eventArgs);
                    break;
                case Event.WriteStarted:
                    WriteStarted?.Invoke(this, eventArgs);
                    break;
                case Event.WriteCompleted:
                    WriteCompleted?.Invoke(this, eventArgs);
                    break;
                case Event.Aborted:
                    Aborted?.Invoke(this, eventArgs);
                    break;
                case Event.Destroyed:
                    Destroyed?.Invoke(this, eventArgs);
                    break;
                case Event.Created:
                    Created?.Invoke(this, eventArgs);
                    break;
                default:
                    throw new NotImplementedException();
            }
        }
```

Maybe my implementation is not correct, find my code below. I run on VS 2019 and in .net framework 4.7.2
```
var myTag1 = new Tag<RealPlcMapper, float>()
  {
  Name = "testtag[0].myTag", //also tried with a tag like "testtag"
  Gateway = "192.0.0.0",
  Path = "1,0",
  PlcType = PlcType.ControlLogix,
  Protocol = Protocol.ab_eip,
  Timeout = TimeSpan.FromSeconds(5)
  };
  
  
  myTag1.Initialize();
  //Write value to PLC
  //This will call Initialize internally since it's the first use of this tag
  //myTag.Value = 3.75;
  //myTag.Write();
  rtxtABAnswer.Text += $"Write myTag" + Environment.NewLine;
  //Read value from PLC
  //Value will also be accessible at myTag.Value
  myTag1.Read() ;
  var result = myTag1.Value;
  rtxtABAnswer.Text += $"myTag value {result}" + Environment.NewLine;
```
I also tried to get all the project with example, compile it and run it on the computer who have access to the PLC but it did not work I have a message "Windows cannot access peripheric, file or path access" ...

Thanks for your help

timyhac, 2022-04-29:
Lets first check that you're able to access the correct underlying libplctag C library.

For a new program, can you please install the libplctag nuget package, and only get the version of libplctag. It should be "2.5.0".
```csharp
var version = $"{LibPlcTag.VersionMajor}.{LibPlcTag.VersionMinor}.{LibPlcTag.VersionPatch}";
```

For the first issue, it looks like you've been able to narrow down where the issue is. Is it possible you could go a little bit further and tell us what the values for `eventTagHandle`, `eventCode`, and `statusCode`. It would be very unusual to get an eventCode that isn't one of those pre-defined.

For the second problem, my best guess is that you aren't able to access the underlying libplctag C library. Can you try the steps listed here: https://github.com/libplctag/libplctag.NET#developing-for-systems-with-immutable-application-directories

Kevin138, 2022-07-08:
I have encountered a similar problem. When I call the Tag.Read() function for my UDT the program breaks and I get the following exception info box.
[image]

Copying the details and pasting into notepad gives me this:
```
System.NotImplementedException
  HResult=0x80004001
  Message=The method or operation is not implemented.
  Source=libplctag
  StackTrace:
   at libplctag.NativeTagWrapper.coreLibEventCallback(Int32 eventTagHandle, Int32 eventCode, Int32 statusCode)
```
I tried to follow instructions for "Developing for systems with immutable application directories" linked above. I downloaded the latest plctag.dll and added it to my project and set "Copy to Output Directory" to "Copy always".  However I am not sure what you mean by this: "Set plctag.ForceExtractLibrary = false before any other calls to libplctag." What is plctag in reference to? The Tag<_UDTMapper, _UDT> variable or something else?

timyhac, 2022-07-08:
> "Set plctag.ForceExtractLibrary = false before any other calls to libplctag." What is plctag in reference to?

In answer to your direct question, that is a line of code to copy and paste into your program. You'll need to add the plctag.NativeImport namespace.

I'm not sure this will help you.  The previous poster had two separate problems, so that suggestion was for the other problem.

johncharris, 2022-07-09:
There was a new version of the libplctag C library released recently that causes a NotImplementedException . By downloading the dll for 2.5.0 and not using the NativeImport package I was able to get it working. @timyhac  posted the link earlier. https://github.com/libplctag/libplctag.NET#developing-for-systems-with-immutable-application-directories

Kevin138, 2022-07-09:
I rolled back to NativeImport v1.0.30 (from 1.0.33). Seems to have solved my issue.

closed 2022-10-05 by timyhac