libplctag/libplctag.NET #253: Reading / Writing to Array members

ga-pc, 2022-05-03:
Hi,

If i try to read or write to a member of an array e.g. PaulTest.Ints[5] I get an error 'ErrorBadParam' as soon as I call Initialize on the tag.

I can work round this for reading, by reading the array from the start and picking out the member of the array that I require, although it is a bit of a pain.
How though, can I write to a single member of an array, without affecting any other values? I can read the array, change the required value and then write it all back, but how do I know that one of those other values has not been changed in the time it takes to do this?

Am I missing something obvious? Surely there must be a way to read from and write to a single member of an array.

I haven't even got to reading/writing a single bit from an int yet....

Thank you.

Paul

timyhac, 2022-05-03:
Yes should be possible, can you supply a minimal example of the code you've tried?

kyle-github, 2022-05-04:
The underlying protocol allows you to access whole tags or parts of tags based on the name you use.   If you use a name like `myArray[10]` you get only array element 10 (or at least elements starting from element 10).

What is the element count you are using?

ga-pc, 2022-05-04:
`            Tag<DintPlcMapper, int> DestinationTag = new Tag<DintPlcMapper, int>()
            {
              Name = "PaulTest.DInt32[12]",
              Gateway = "172.16.80.181",
              Path = "1,0",
              PlcType = PlcType.ControlLogix,
              Protocol = Protocol.ab_eip,
              Timeout = TimeSpan.FromMilliseconds(500)
            };
            DestinationTag.Initialize();
            DestinationTag.Read();
`

This is the code that is failing and giving me the ErrorBadParam result.
It throws an exception as soon as it tries to initialize the tag.

I am not setting the element count, so that presumably defaults to 1.
I can read discrete tags with this code perfectly  (i.e. those not in an array)

Thank you

timyhac, 2022-09-29:
Hi @ga-pc apologies for leaving this one so long. Were you able to resolve the issue?

closed 2022-10-05 by timyhac