libplctag/libplctag.NET #257: Multiple Requests Per Packet

airwedge1, 2022-05-10:
I know the underlying Libplc library supports multiple requests per packet for controllogix / compact logix.  Is there a way to do this in the .net wrapper?  For reference: https://github.com/libplctag/libplctag/issues/15

timyhac, 2022-05-10:
While we don't have explicit control over how many requests per packet there are, if you're simply looking for concurrent requests just do something like

Task.WhenAll(tagA.ReadAsync, tagB.WriteAsync)

airwedge1, 2022-05-10:
I would think that would just submit 2 requests in parallels vs 2 requests within the same packet.  Or is it smart enough to combine the requests together?

timyhac, 2022-05-10:
It is smart enough to combine the requests together

closed 2022-05-10 by airwedge1

kyle-github, 2022-05-10:
The core library will combine requests unless you serialize all your access and wait for each request to complete before starting a new one.