libplctag/libplctag.NET #259: Visual Studio->Publish->Produce single file

marioan, 2022-05-21:
First, thank you for your library, it is very good and works well for me.
I am working with Winodws, Visual Studio 2019, C#, Net5.0, Asp.net core.

I was trying to Puplish the application with the Produce Single File Options, but no 
success , It is not working.
I know that probably is because something is wrong with the libplctag.NativeImport library that fail to write the plctag.dll ...
I read something about that in  in https://github.com/libplctag/libplctag.NET/blob/master/docs/Using-a-non-packaged-version-of-the-native-libplctag-library.md

My question is very simple, there is a simple way to use only libplctag without 
libplctag.Nativeimport ? I will put the right plctag.dll in the Project.

Any advice to help me to use Single File Options ?
Thank you again

Mario Andreis

jkoplo, 2022-05-21:
I've tried to use the "single file" output too without a lot of success. In theory it should be possible...

In order to make things easier with libplctag, the native library extracts the necessary base DLL to disk and then dynamically links to it at runtime. This makes sure that the right OS/arch DLL is loaded, etc.

Unfortunately, there are so many different OS and deployment variations that things get complicated. Sometimes the directory the exe is in is write protected, etc, etc.

In the case of the single file output, Microsoft does some really weird things with paths. And to make it harder, there was a major change in how the single file output was created and how the paths for the exe worked.

We've tried to make libplctag.net seamless to use, but there are a bunch of corner cases with loading the base library. I think we could provide an "escape hatch" of sorts if we allowed end users to load libplctag.Native by passing a path to the base DLL they want to use. Right now you can tell Native to not extract the DLL, but it still assumes the path to load from.

@timyhac, thoughts?

marioan, 2022-05-23:
Thank you Jkoplo for your reply

I have made some trials just to understand what's happen.
With 'single file' option the  libplctag.NativeImport does not write/rewrite plctag.dll, this should happen every time that the application starts.
That is not a big problem for me, because I simply add the right OS/Arch plctag.dll to my project, and when publish the plctag.dll is in the same folder with the application.

That said, In my opinion the problem is because the libplctag.dll /libplctagWrapper.dll that are embedded in the single file application can not reach the plctag.dll

Probably you already know about that, but there are some API compatibility issues with the way to discover the Application folder
https://docs.microsoft.com/en-us/dotnet/core/deploying/single-file/overview

[image]

thanks again
Mario Andreis

jkoplo, 2022-05-23:
Yeah, and where all different path options (appcontext, environment, etc) point has changed across dotnet core versions...

That's why I think we should allow an override where the user figures out their path to the DLL outside of the library and just tells the library where it is.

marioan, 2022-05-23:
Yes
Write somewhere (in text file format, for example .txt) which is the application path (That is the same where to find plctag.dll) would be nice to have.

Thank you again for your time, 
and let us know if/when you implement this feature

Ciao
Mario Andreis

marioan, 2022-05-25:
Jkoplo

I was able to run my application with the 'single file' option.

1-create a new libplctag.NativeImport dll with 
static private bool _forceExtractLibrary = false; and 
static private bool libraryAlreadyInitialized = true;

[image]

2-link libplctag and libplctag.NativeImport as assemblies (No Packages)

[image]

3-copy the right OS/arch plctag.dll in the application folder

and it works
thank you again for your time

Mario Andreis

jkoplo, 2022-05-25:
That's a decent work around and proves some stuff out. I'd still like to implement a proper fix, but not sure when I'll find some free time for it.

@timyhac let's leave this issue open so I can eventually PR against it.

marioan, 2022-05-25:
could be that the problem is inside LibraryExtractor.Init(ForceExtractLibrary); ?
Never return from this call because an internal error!

[image]

timyhac, 2022-06-21:
The other idea to pursue is the fix for https://github.com/libplctag/libplctag.NET/issues/137 - maybe we can produce an alpha build of https://github.com/libplctag/libplctag.NET/pull/249 and see if it works?

timyhac, 2024-07-17:
I have released an alpha version of libplctag.NativeImport which contains the fix for #137 and have just tried it with PublishSingleFile and SelfContained and it seems to work.

`Program.cs`
```csharp
using libplctag;
var result = LibPlcTag.IsRequiredVersion(2, 6, 0);   // This will crash if it can't reach the native libraries.
Console.WriteLine(result);
Console.ReadKey();
```

`libplctag_single_file_test.csproj`
```xml
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>net8.0</TargetFramework>
    <ImplicitUsings>enable</ImplicitUsings>
    <Nullable>enable</Nullable>
    <PublishSingleFile>true</PublishSingleFile>
    <SelfContained>true</SelfContained>
    <RuntimeIdentifier>win-x64</RuntimeIdentifier>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="libplctag" Version="1.5.2" />
    <PackageReference Include="libplctag.NativeImport" Version="2.0.0-alpha.1" />
  </ItemGroup>

</Project>
```

View of the publish folder, and the program running in a terminal:
[image]

@marioan - sorry that this is more than 2 years late, but I think this is the path forward!

marioan, 2024-07-17:
Thank you Timyhac
I am still working with LibPlcTag
I have to make a new installation soon, and I will try this new version
Thanks again

timyhac, 2024-07-17:
Great! If you do have problems please a ticket and I'll do what I can to help.

closed 2024-07-17 by timyhac