libplctag/libplctag.NET #263: Polling/Subscription Wrapper

MarcBertrand, 2022-08-09:
Kinda new to the Git-world so lemme know if this is the wrong place to post this.  Wondering if anyone has done a polling/subscription oriented wrapper for this.  Something that works more like an OPC subscription where polling is handled in the wrapper and notifies on data changes.  Looking at writing one but don't want to re-invent the wheel.

jkoplo, 2022-08-09:
I have sort-of, kind-of. 

I've done it internally for work and also in another open source project.

What's your end goal? I might be able to add some example code to the repo or the you to my other project.

MarcBertrand, 2022-08-10:
End goal is to be able to avoid use of drivers like Kepware.  Currently using OPC with a subscription model.  What I am thinking is to emulate OPC behaviors.  In OPC there are methods for read, write, subscribe.  A subscription has an update/polling interval.

So, if I subscribe a group of tags with a 1000ms interval, the driver would execute reads every 1000ms but also compare the new value to the last value and notify with a callback of any value changes.

Any example code you could share would be appreciated.  I will share out my finished product with any enhancements.  I think having a polling model will be a good addition.

Marc Bertrand
Sage Clarity
+1 636-226-7427

kyle-github, 2022-08-11:
Sorry for the email reply, I am on vacation with limited Internet access.

The core library can support a subscription model but you need to do a
little work.  You need to create a tag handle with the automatic read
attribute and with a callback.

A wrapper like the .Net one could probably make this look a lot nicer.

There was a user who mentioned creating a version of an open source OPC
server that used libplctag for CIP.  I don't think the work was ever
released publicly.

Best,
Kyle

MarcBertrand, 2022-08-15:
Hey Kyle,

You seem to know this library pretty well.  Any chance you or someone you know from the OSS project might be interested in picking up some side work?

My DevOps team is not well suited for something like this.  I come from a PLC integration background so I could probably figure it out but I have time constraints.  I’ve engaged specialized development from places like Upwork before and have budget for this kind of thing.

I’d love to get this embedded in our stack.  Trying to figure out the best way to get going.

Thanks,

Marc

Marc Bertrand
Sage Clarity
+1 636-226-7427

jkoplo, 2022-08-15:
@MarcBertrand - What sort of 'callback' are you looking for? Just a simple event from a library in C# or some sort of webhook? Or are you running an OPC-UA client on your end and looking for a special OPC-UA message back?

The OPC-UA library for .NET ([OPC UA .NET](https://opcfoundation.github.io/UA-.NETStandard/)) is basically the reference implementation and we use it professionally to host OPC-UA servers within our code. Tying it together with libplctag.NET isn't terribly hard, but as far as I know no one has published that in an open source version. That would basically just be an open source OPC-UA server that could target AB PLCs (and a few others).

I've considered slapping the two together myself and publishing it, but I haven't gotten into the depths of OPC-UA yet. I'm working on another project ([CrudeObservatory](https://github.com/jkoplo/CrudeObservatory)) that was originally a configurable router between PLCs and storage formats/endpoints, but is starting to look more and more like an open source Historian server.

It sounds like you are just looking to replace Kepware with something that doesn't have a license fee?

kyle-github, 2022-08-16:
@MarcBertrand I can't do any contract work due to my contract with my employer.  That's how I keep libplctag independent.   I think the C# OPC/UA implementation that Jody mentions would be a good starting place if you really need OPC.  If that is not what you need, then asking on the [forum](https://groups.google.com/g/libplctag) might yield some results.

timyhac, 2022-09-14:
Hi @MarcBertrand - I believe this issue can be closed?

MarcBertrand, 2022-09-14:
Yes, thank you!

Marc Bertrand
Sage Clarity
+1 636-226-7427

closed 2022-09-14 by timyhac