libplctag/libplctag.NET #264: TagInfo for non program tags

MilDur, 2022-08-23:
Hello,
   Im facing issues related to reading taginfo for non-program tags on lower level than "root" . The reading works fine but without knowing the type its little bit tricky select the right plcmapper.

Thanks for any advice.

timyhac, 2022-08-23:
Can you bit a bit more specific around what the issue is?

Listing tags will return a sequence of TagInfo which should provide the tag data type code: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ExampleListTags.cs

MilDur, 2022-08-23:
Sure.
  I would like to get taginfor for tag "Line.Id" or be able to list all taginfo in "Line.@tags" (not Program:Line.@tags). im still getting ErrorOutOfBounds exception. 
Thanks for suggestions

kyle-github, 2022-08-23:
Are you trying to get tag info about fields within a UDT?  If so, the protocol does not appear to support that.   I've tried.

The only things that I have seen work are controller tags (root) and program tags.

MilDur, 2022-08-25:
yes, exactly trying to browse/scan udt. i managed to browse with udtinfoplcmapper. i will continue to simplify the full plc scan to achieve what i want. get info if udt exists and what type it is.

kyle-github, 2022-08-25:
There are only two protocol commands that let you get data about tags.  The first lists all tags (actually it lets you iterate over all instances of the tag/symbol class).  The second lists all info about a single UDT.

The first uses the "@tags" special tag.   The second uses the "@udt/xxxx" special tag.

There is no way I have found of listing out information for a single tag.

The only way I know of getting information about a single UDT tag is to list all tags, find the tag you want, look at the UDT ID number, and then look up the info for that UDT. 

UDT IDs are only 12 bits so there can only be about 4096 of them.   Tag IDs are 16 bits so there can only be about 64k of them.

MilDur, 2022-09-06:
Ive managed to get info about any of UDTs by browsing (listing) all tags from from root level to the tag itself. So I feel this issue can be closed. Thanks to all for your suggestions and tips.

closed 2022-09-06 by MilDur