libplctag/libplctag.NET #265: Timeout issue

patopat, 2022-08-25:
Hi,

I run the libplctag.NET sample ILoggerExample with modified name,gateway and Path in the host machine.
I configured in a VM an emulator test project :
[image]
[image]

[image]

[image]

it seems connection ok, but something is wrong, could you help me ?

Here is the log window :

```cmd
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.212 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.216 thread(1) tag(0) INFO initialize_modules:177 Starting.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.217 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.218 thread(1) tag(0) DETAIL mutex_create:726 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.221 thread(1) tag(0) DETAIL mutex_create:753 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.222 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.223 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.224 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.225 thread(1) tag(0) INFO hashtable_create:75 Starting
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.227 thread(1) tag(0) INFO hashtable_create:99 Done
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.242 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.262 thread(1) tag(0) DETAIL mutex_create:726 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.468 thread(1) tag(0) DETAIL mutex_create:753 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.592 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.710 thread(1) tag(0) DETAIL cond_create:1087 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.768 thread(1) tag(0) DETAIL cond_create:1114 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.880 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.895 thread(1) tag(0) DETAIL thread_create:884 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.902 thread(1) tag(0) DETAIL thread_create:917 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.921 thread(1) tag(0) INFO lib_init:161 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.916 thread(2) tag(0) INFO tag_tickler_func:508 Starting.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:36.968 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:37.009 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:37.016 thread(1) tag(0) DETAIL mutex_create:726 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:37.021 thread(1) tag(0) DETAIL mutex_create:753 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:37.024 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
[... 744 lines cut ...]
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.528 thread(3) tag(0) DETAIL session_handler:1237 in SESSION_WAIT_RETRY state.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.550 thread(2) tag(11) DETAIL tag_tickler_func:545 Tickling tag 11.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.671 thread(2) tag(11) DETAIL plc_tag_generic_tickler:276 Tickling tag 11.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.671 thread(3) tag(0) DETAIL session_handler:1237 in SESSION_WAIT_RETRY state.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.732 thread(2) tag(11) DETAIL plc_tag_generic_tickler:381 Done.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.765 thread(2) tag(0) DETAIL tag_tickler_func:631 Tag tickler thread timed out waiting for something to do.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.778 thread(3) tag(0) DETAIL session_handler:1237 in SESSION_WAIT_RETRY state.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.780 thread(2) tag(11) DETAIL tag_tickler_func:545 Tickling tag 11.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.809 thread(2) tag(11) DETAIL plc_tag_generic_tickler:276 Tickling tag 11.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-25 11:23:42.849 thread(2) tag(11) DETAIL plc_tag_generic_tickler:381 Done.

```

Best regards,
Pat

kyle-github, 2022-08-25:
People have not had good luck with the emulator.   That said, your path is incorrect.  It should be "1,2".  The first element is the port from the network module.  In this case "1" means the back plane of the virtual chassis.   Then the second element is the slot of the CPU.   That would be "2".

patopat, 2022-08-29:
@kyle-github,

Ok, I changed the path like this :
```Csharp
    var myTag = new Tag<DintPlcMapper, int>()
    {
        //Name = "MyTag[0]",
        //Gateway = "127.0.0.1",
        //Path = "1,0",
        Name = "EmulatorTestProject.Test",
        Gateway = "192.168.80.130",
        Path = "1,2",
        PlcType = PlcType.ControlLogix,
        Protocol = Protocol.ab_eip,
        Timeout = TimeSpan.FromSeconds(20),
    };
```

Could you help me to find what exactly block the connection ? I guess it's when "warn: " messages are coming but I can't interpret what exactly is the meaning.

```CMD
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.803 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.805 thread(1) tag(0) INFO initialize_modules:177 Starting.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.806 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.808 thread(1) tag(0) DETAIL mutex_create:726 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.809 thread(1) tag(0) DETAIL mutex_create:753 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.810 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.811 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.812 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.813 thread(1) tag(0) INFO hashtable_create:75 Starting
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.815 thread(1) tag(0) INFO hashtable_create:99 Done
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.815 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.817 thread(1) tag(0) DETAIL mutex_create:726 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.818 thread(1) tag(0) DETAIL mutex_create:753 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.820 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.820 thread(1) tag(0) DETAIL cond_create:1087 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.822 thread(1) tag(0) DETAIL cond_create:1114 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.822 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.824 thread(1) tag(0) DETAIL thread_create:884 Starting.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.825 thread(1) tag(0) DETAIL thread_create:917 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.826 thread(1) tag(0) INFO lib_init:161 Done.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.826 thread(2) tag(0) INFO tag_tickler_func:508 Starting.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.826 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:44.829 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
[... 368 lines cut ...]
      2022-08-29 13:31:45.188 thread(3) tag(0) INFO recv_eip_response:1998  02 00 00 00 00 00 b2 00 04 00
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.189 thread(3) tag(0) INFO recv_eip_response:1998  db 00 08 00
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.190 thread(3) tag(0) INFO recv_eip_response:2005 Done.
warn: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.192 thread(3) tag(0) WARN receive_forward_open_response:2326 Forward Open command failed, response code: Unsupported service. (8)
warn: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.226 thread(3) tag(0) WARN receive_forward_open_response:2329 Received CIP command unsupported error from the PLC!
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.228 thread(3) tag(0) INFO receive_forward_open_response:2369 Done.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.229 thread(3) tag(0) DETAIL session_handler:1126 PLC does not support ForwardOpenEx, trying old ForwardOpen.
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.230 thread(3) tag(0) DETAIL session_handler:1102 in SESSION_SEND_FORWARD_OPEN state.
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.231 thread(3) tag(0) INFO send_forward_open_request:2043 Starting
dbug: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 13:31:45.232 thread(3) tag(0) DETAIL send_forward_open_request:2093 Set maximum payload size guess to 508 bytes.

MANUALLY REMOVED NEXT LINES
```

Regards,
Pat

kyle-github, 2022-08-29:
Hi Pat,

Thanks for the log dump.  Unfortunately, it was cut off at a normal point.  The library tries the newer extended CIP connection opening handshake command and if that does not succeed drops back to the older one.   The last thing we see happen above is that the newer command is returning a "not supported" error from the PLC, so the library will drop back, but that is in the log after this point.   The WARN messages are completely normal.  I should probably change the log status to INFO for those as they are not error conditions.

Could you attach the full log?

patopat, 2022-08-29:
Hi @kyle-github,

Yes :) 
[cmd.log](https://github.com/libplctag/libplctag.NET/files/9445380/cmd.log)

kyle-github, 2022-08-29:
Here's the relevant part:

```text
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 16:24:36.868 thread(3) tag(0) INFO recv_eip_response:1998  d4 00 01 01 09 01 9f 5d 3d f3
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 16:24:36.870 thread(3) tag(0) INFO recv_eip_response:1998  45 43 50 21 02 00
info: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 16:24:36.874 thread(3) tag(0) INFO recv_eip_response:2005 Done.
warn: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 16:24:36.877 thread(3) tag(0) WARN receive_forward_open_response:2326 Forward Open command failed, response code: Invalid Connection Size (1)
warn: CSharpDotNetCore.LibPlcTagLogSource[0]
      2022-08-29 16:24:36.879 thread(3) tag(0) WARN receive_forward_open_response:2351 CIP error code Invalid Connection Size (This error occurs when the target device doesn't support the requested connection size. Check the documentation of the manufacturer's device to verify the correct Connection size required by the device. Note that most devices specify this value in terms of bytes. The Productivity Suite CPU does not return this error and will instead use errors 0x0126, 0x0127 and 0x0128.)!
```

Interesting that this is the error we get.   Not what I expected.   The default is correct for a Control/CompactLogix, 544 bytes.   But that isn't correct for the Emulator.  I do not see any hints about the size in the returned packet.  The remaining bytes are just a copy of the some of the connection set up info.  

Generally you can request a smaller size than the maximum allowed, so the size needed is probably less than 544.   When connecting to an older PLC such as a PLC/5, 508 is used.  So when this error is found, it usually means that the requested connection size is too big.

I do not currently allow the tag string to specify the attempted connection size.

I will need to think about what to do here.   Using the Emulator is something that others have wanted and you seem to be getting a lot farther than people have in the past.

patopat, 2022-08-29:
@kyle-github ,
Thank you very much for your help.
Let me know if you need that I test some code to validate idea or anything else to help  :)

Regards,
Pat

aTunell, 2022-08-30:
I was able to get this to work using FactoryTalk Logix Echo with the following command .\list_tags_logix.exe 127.0.0.2 1,0

Logix Echo is the new emulator for Rockwell PLCs that is only compatible with Firmware v33 and forward. I would be astounded if you could get the old emulator (Logix Emulate) to work, given that it uses RSLinx Classic, which doesn't assign the emulated PLC an IP address.

patopat, 2022-08-30:
@aTunell ,

Could you confirm that is "FactoryTalk Logix Echo ControlLogix 5580" ? I can see a single node product also...

Regards,
Pat

aTunell, 2022-08-30:
I'm using Logix Echo Single Node. I'm not sure what the ControlLogix 5580 product is. I believe it may be an exclusive emulator for L8 controllers. Fact is, Logix Echo currently only supports L8 controllers anyway, but they're working on GuardLogix and Compact Logix for future releases. 
Logix Echo is currently at 1.0, and has some curious bugs, including not releasing floating licenses when you shut it down, so be careful.

patopat, 2022-09-03:
Hi @kyle-github , @jkoplo , @aTunell 

Good news 😃 !!
I received the FT Echo and I successfully run the example `ExampleListTags` from LibPlcTag.Net in the VM where RA tools are installed.
The next step is to run the example from my host machine to communicate via the VM...

[image]

Here is the changes made :

```Csharp
            var tags = new Tag<TagInfoPlcMapper, TagInfo[]>()
            {
                //Gateway = "192.168.0.10",
                Gateway = "127.0.0.2",
                Path = "1,0",
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Name = "@tags"
            };
```
FT Echo config :

[image]

Logix Designer config :

[image]

Regards,
Pat

kyle-github, 2022-09-03:
It is really good to hear that the newer emulator works.   I just cannot comprehend why the emulator would be so broken before.   It is their own code!

I would like to continue to see how far we can get with the older emulator.   I only have Studio 5000 v31.  From the logs above, it looks like it is really close to working.

patopat, 2022-09-04:
@kyle-github ,
The libplctag program is running successfully from my host machine and the VM where RA tools are installed :)

I keep the VM with Studio 5000 V32.04 and Emulate for some test.

Regards,
Pat

timyhac, 2022-09-29:
Hi @patopat @jkoplo  @kyle-github - looks like this one has been resolved, OK to close or is there an action on documenting emulator usage?

closed 2022-10-05 by timyhac