libplctag/libplctag.NET #266: How to read a part of a Struct ?

patopat, 2022-09-14:
Hi,

I was wondering if it is possible to read 2 consecutive float from a specified name and a given length in order to have one read access like :

 readTag   Name=myTest[1].x2    length=8 bytes ?

test_DUT  myTest[2];

Struct test_DUT{
 float x1;
 float x2;
 float x3;
}

Regards,
Pat

timyhac, 2022-09-14:
Correct me if I'm wrong, but I think you're trying to read an array of structs, you only care about one field within the struct - and you only want to call Read once.

In this case, you will need to read the array, and then pull out what you require.
I would recommend studying the example mapper code here: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/SequencePlcMapper.cs
https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ExampleCustomUdt.cs

Depending on what you're really trying to to achieve (e.g. performance???) you may look at using the async versions of the methods and call them concurrently. Under the hood, libplctag will pack the network requests together so it is more efficient over the wire.

patopat, 2022-09-14:
Hi @timyhac ,

Yes I would like read 2 members via a given path. In one call, read x2 and x3 with the path myTest[1].x2  and a length of 8bytes.
Yes, I successfully try the mapper of my custom DUT. You are right, I would like optimize.
Please, could you elaborate a little bit the async version with a short example ?
Regards,
Pat

timyhac, 2022-09-14:
There are many examples here https://github.com/libplctag/libplctag.NET/tree/master/src/Examples/CSharp%20DotNetCore. In particular the ExampleAsync class has some variations for you to study.

timyhac, 2022-09-29:
Hi @patopat - I trust that your question has been answered?

closed 2022-10-05 by timyhac