libplctag/libplctag.NET #270: Trying to read high frequency (100Hz)

kbeaugrand, 2022-09-18:
Hi, 

I'm trying to use this library to poll some tag in high frequency (around 100Hz) on a ControlLogix.
It's certain that my tag values changed every 10ms but I miss some changes when I'm reading that.

The read operation takes 4-5ms, but after some tries, it seems that the tag kept the old values (cache somewhere) ?
I verified that ``ReadCacheMillisecondDuration`` was set to 1 (hopping to disable the cache).

Could you tell me what I'm suppose to verify in my code to ensure that I can read as fast as possible using this library ? 

Regards,

jkoplo, 2022-09-18:
10ms is very fast for most of the AB controllers - it's right on the edge of doable if you're only reading a small number of tags and there aren't other network operations (HMI/SCADA) hitting the PLC. A lot of this comes down to the network stack and hardware on the PLC itself. There is a model of AB that has a gigabit network interface and you may be able to get those kinds of rates out of that PLC - I've never had access to one for testing.

I've got some notes and I've done some benchmarking on max read speeds. We often get questions about it, so I should probably gather it all together in one place and post it as a doc or wiki.

kbeaugrand, 2022-09-19:
@jkoplo , 

Thank you for your quick response.

I was indeed afraid that this would not be feasible. The idea would be to get about 30 tags (DINT) at this frequency.

I also tried to retrieve an entire array (which contains these tags) but it seems to take between 100 and 700ms to make the call, instantiate the resources and map them to memory...

So I think I'll reach a dead end to do this, but it's always interesting to know what limits we can expect...

If you make a documentation about your tests and their implementation I'll be very interested to read it !

closed 2022-09-19 by kbeaugrand

kbeaugrand, 2022-09-20:
> Yeah, 10 msec is even getting down to the point where the scan rate of the PLC is going to be questionable, depending on model.
> 
> With that said, you could have some scheme where you load the values into an array as a buffer and read it at a slower interval. Reading an entire array is more efficient based on how CIP packs tag names and values into packets. This would add complexity in terms of knowing which items in the buffer are stale, etc.
> 
> Frankly I suspect that some dedicated data acq hardware might be the ideal solution here...

Honestly, since reading a small array at this frequency is taking more time than the update frequency, I think that the buffer will be updated more that I can read it...
The dedicated hardware should be the only one solution for my case I guess.

However, @jkoplo , regarding my case I'm also trying to read an unique DINT that changes every 400ms. I tried to poll its value every 100/200/400ms and it seems that I still miss some values even if the read operation was successfull.

For exemple. My value is incremented by a power of 2 every 400ms (1,2,4,8,16,32...) when reaching 512, it will restart to 1.
When polling every 100ms I had this kind if results : 

<html>
<body>
<div><br/><br/>

timestamp | TAGVALUE
-- | --
2022-09-20T08:52:25.5800285Z | 4
2022-09-20T08:52:25.4802025Z | 256
2022-09-20T08:52:25.3797306Z | 256
2022-09-20T08:52:25.280438Z | 256
2022-09-20T08:52:25.1801403Z | 256
2022-09-20T08:52:25.0803685Z | 64
2022-09-20T08:52:24.979968Z | 64
2022-09-20T08:52:24.8805511Z | 64
2022-09-20T08:52:24.7801575Z | 64
2022-09-20T08:52:24.6797418Z | 16
2022-09-20T08:52:24.5815462Z | 16
2022-09-20T08:52:24.4796859Z | 16
2022-09-20T08:52:24.3803935Z | 16
2022-09-20T08:52:24.2800734Z | 4
2022-09-20T08:52:24.1800835Z | 4
2022-09-20T08:52:24.0801315Z | 4

</div>
</body>
</html>

How can we explain that the library returns a value that should have change in the PLC ? Could it be due to the PLC scan duration ?
After looking deeply, some times I get the missing values (1,2,8,32)...

kyle-github, 2022-09-20:
You shouldn't have any trouble reading at that rate.  What kind of PLC is it and what kind of network module does it have?   Even on my terrible WiFi network, I can read consistently faster than every 100ms.

The pattern you show above is odd.   It is multiplying by 4, not 2 each time and then seems to keep the same value for the right amount of time.   You might want to try slowing it down a bit and watching the value in Studio 5000 to see if it is really getting all the different values.

Are you using read caching?

kbeaugrand, 2022-09-20:
@kyle-github , 

I see sometimes the expected values. The program is a simulation one that should be only in charge to increment these values. I'm not an automation expert, I'll ask my colleagues to troubleshoot on the PLC to verify that.

Concerning the read caching, I'm reading the tags by this way : 

```csharp
            var tag = new Tag<DintPlcMapper, int>();
            tag.Name = tagDefinition.TagName;
            tag.Gateway = tagDefinition.Gateway;
            tag.Path = tagDefinition.Path;
            tag.PlcType = tagDefinition.PlcType;
            tag.Protocol = Protocol.ab_eip;
            tag.Timeout = timeout ?? TimeSpan.FromSeconds(1);
            tag.ReadCacheMillisecondDuration = 0;

            plcTag.Read();
``` 

Maybe I miss something ? or something wrong in my sample ?

kyle-github, 2022-09-22:
Are you open/creating the tag handle, doing a read, and then closing it?   Opening and closing the connection to the PLC is a very slow and heavy operation.

kyle-github, 2022-09-22:
Also note that if you use an AB PLC with a 1Gbps network link (L8x series), and you have a good wired network (all 1Gbps) you can see read times of around 1-2ms in many cases.   At least one user is getting those kind of speeds.   

Generally it will be hard to hit 100Hz.   If nothing else, the PLC will be working very hard to handle all of your requests.  As Jody said above, log the data into an array and then read a bunch of it at a time.  I have done that in the past and it doesn't take much of a buffer to make it feasible and drop the load on the PLC.

kbeaugrand, 2022-09-22:
> Are you open/creating the tag handle, doing a read, and then closing it? Opening and closing the connection to the PLC is a very slow and heavy operation.

I instantiate the tag before entering the loop, and close/dispose of it when exiting the loop. The first call may take some time (this is not a big challenge).
This seems to work as I can use my data and see that I have more or less what I expected: 

[image]

This is not a surprise, but we can see that the TimerTrigger is not perfect. Some executions need some ms (between 1 and 5ms) to be triggered, and this is not really a value every 10ms...

airwedge1, 2022-11-22:
Just randomly saw this issue.  Assuming you are doing this for data collection, we've had to modify the ladder logic to store off the values in an array on the PLC at the frequency and round robin it.  You can store a timestamp + value and then you can get many values from the arrays much quicker then getting the same value many times.