libplctag/libplctag.NET #275: How to retrieve tag values in a Read Complete event callback?

PWSys, 2022-10-02:
When using the Async Read Complete event callback. What is the best way to get the value returned?

With the code below, the object being passed is of type NativeTagWrapper and doesn't seem to expose a value attribute. TagEventArgs is only returning "Ok" for status.   
 
```
var testTag = new Tag<DintPlcMapper, int>
{
    Gateway = plc.IPAddress,
    Path = plc.Path,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Name = "testDINT"
};
testTag.ReadCompleted += TestTag_ReadCompleted;
testTag.InitializeAsync();
testTag.ReadAsync();

await Task.Delay(10000);
```

` void TestTag_ReadCompleted(object? sender, TagEventArgs e)
{
    Console.WriteLine($Read Completed");
}`

timyhac, 2022-10-02:
@PWSys - its not a direct answer to your question but is it possible for you to await the ReadAsync call and then access the value directly (i.e. on line 12 of your example).

i.e.
```
var testTag = new Tag<DintPlcMapper, int> ...

await testTag.ReadAsync();
Console.WriteLine($"Read Completed. Value = {testTag.Value}");
```

If not, you could declare an inline delegate to pass the reference through?
```
testTag.ReadCompleted += (sender, e) => ReadCompleted(testTag, e);
ReadCompleted(object? sender, TagEventArgs e) => Console.Writeline($"Read Completed. Value = {sender.Value}");
```

Having said that, I admit that exposing the NativeTagWrapper through the sender parameter is a flaw in the API design. THe class itself is private so shouldn't really be accessible to end-users.

PWSys, 2022-10-02:
Thanks @timyhac !!

Using the delegate worked!  I also pulled in an updated copy of the library with #276 and it too worked as expected.  Thanks again! 

I did notice something else and I'm not sure if it's related.  When using AutoSyncReadInterval the value in the ReadComplete event callback is static with TagEventArgs indicating "Ok".  The callback complete event fires in the timespan expected, it's just that the value never changes.

The only time this value updates is when explicitly calling ReadAsync.  Then the AutoSyncReadInterval will indicate the updated value.  From this point on, it will be static until calling ReadAsync again.

Is this the expected behavior when using AutoSyncReadInterval?

timyhac, 2022-10-02:
I see... I think no, you wouldn't expect that, you would expect the updated value to be available on the Value property 

Fixing this for read should be straightforward - we can just call DecodeAll on a WriteCompleted event, but fixing it for auto-Write won't be - at least in general.

PWSys, 2022-10-03:
Wow, thank you @timyhac !  I'll give this a try later today.

closed 2022-10-03 by PWSys

PWSys, 2022-10-03:
@timyhac , I tested this change, and it works as expected.  Thanks again for your help!