libplctag/libplctag.NET #279: AutoSyncWriteInterval does not work for `TagOfT`

timyhac, 2022-10-02:
As part of the discussion on #275 it was discovered that the AutoSync feature does not work for `TagOfT` tags.
The auto-read feature was fixed in #278 but the auto-write still does not work.

Options:
1. Remove access to the auto-sync feature for `TagOfT` (forcing consumers to implement their own autoSync logic).
2. Add a call to EncodeAll on the setter for Value (however this would not work for arrays as they do not access the setter).
3. Expose EncodeAll to consumers and request that they call this after making a change (not ideal, and we want to hide the Mapping logic to consumers).

timyhac, 2022-10-04:
We could trigger EncodeAll when the users sets `Value` (ie in the Setter), which I think is what you're suggesting (please clarify if this isn't the case).

This would work for tags that refer to primitive values (DINT, INT, STRING) etc. But what about UDTs and Arrays? 
e.g. If you were to set the value for a UDT field
 how would TagOfT know?

 ```myTag.Value.MyUdtField = 5```

jkoplo, 2022-11-03:
I'm curious what @kyle-github current thoughts are on some of the auto-sync features.

This might be a bit uninformed, but I think I'd personally prefer to keep as much thread handling and such up in native dotnet. It feels like the more threading we're managing between c and dotnet the more complexity and corner-cases we're asking for.

kyle-github, 2022-11-03:
If you end up calling _any_ of the C DLL data setter functions, the write will be triggered (if it is an auto-write tag).  Does that happen when you write the value like that UDT example?  If you are not encoding the value back into the underlying tag buffer, then nothing would trigger the write.  

For the trigger, it does not matter if you write one byte in a larger tag object, the write will still be triggered.  Any write will do it.   And, it isn't smart.  You could write the same value that is already there and it will still trigger the write.

Also, writes win if there is a write pending and the read time comes up.   This is to ensure that writes always get pushed out to the PLC so that a read does not overwrite the new value.

There is a small delay for write coalescing.   Writes are not triggered immediately. It is pretty important to delay write a bit because some types (like strings) might be written in two passes.

It could make sense to simply implement all the automation for auto-sync in the wrapper.   It would be easier to fully integrate that into the framework of something like .Net, but you'd have to implement properties for all tag fields/elements (for UDTs/arrays) and your own timed events to trigger reads and delayed writes.  

Alternatively you could implement a slightly simpler "sync" idea.   Periodically have a timer event trigger a sync.  If the tag is dirty (something was changed), then do a write.   Otherwise do a read.  

There is some tricky logic.   For instance, if I have a read in flight and someone changes a value in a tag, I have to cancel the read and trigger a delayed write.  There is probably still some sort of race condition in there, but I tried pretty hard to even kill off reads that had succeeded but had not yet processed the data from the PLC.  Once you start processing read data, the internal mutex is locked and the data setters are blocked.

Auto-sync was a _lot_ harder to get right than I thought when I first starting thinking about it.

kyle-github, 2022-11-04:
I had a thought.   I could expose a "mark dirty" function.   Then you could maintain all the tag data in the native format (i.e. .Net) and when someone changes something there, you call the "mark dirty" function.  We'll call it plc_tag_mark_dirty() for now.   Calling this will start the internal auto-sync write machinery.  Mostly that just sets a timer for sometime in the future to trigger a write.

Just before a write is done, there is an event/callback call that is emitted, PLCTAG_EVENT_WRITE_STARTED.   This notifies your the event callback than a write is about to happen.   At that point, you can call the data functions and encode your native data back into the C DLL's buffer.   When the callback returns control back into the C DLL, the changed buffer will be written and you'll eventually get a PLCTAG_EVENT_WRITE_COMPLETED event.

That way, you would be in control of all the triggers for writes.   And it will still interoperate with the auto-read if you want that.

Alternatively you could just skip it all and do it yourself.

jkoplo, 2022-11-04:
I think I like your idea.

I figured there was complexity in the threading and locking with the auto sync stuff - probably best we don't duplicate that if we can help it.

I need to think through the handshake of trigger-wait-event, but it seems clean. Worth prototyping in a branch?

timyhac, 2022-11-04:
How do we know when to call mark-dirty? If we know when to call this we could call the Setter functions. The only benefit I see is that we'd only need to do it once per plc write as opposed to once per write .net object write.

kyle-github, 2022-11-04:
There is a delay between marking a tag dirty and when the write actually triggers.   This is to coalesce changes.  That delay is controlled by the `auto_sync_write_ms` tag attribute.  It is poorly named and should have been something like `auto_sync_write_delay_ms`.

You could just call the setter function too.   It is the difference between eagerly changing (calling the setter immediately) and lazy changing (call the mark dirty function but not calling the setters until you know the write is starting).

The only reason I can see to be lazy is if you want to coalesce changes to the same value before the write happens or the encoding is expensive and you want to only do it at the last minute.

kyle-github, 2022-11-04:
One point about rolling this all yourself, which is perfectly possible, is writes blocking reads.   The existing system makes sure that if a write is flagged for the future (i.e. the tag is "dirty"), then automatic reads are suppressed until the write completes.   This is really important to do because otherwise the incoming read will overwrite your changed data.

So if you want to write this all in the wrapper, please remember this.   I have done a lot of testing to make sure that the existing system works and it was a lot trickier to get this all correct than I thought it was going to be.   If I knew how invasive auto-sync was going to be I probably would not have done it!

timyhac, 2022-11-08:
Out of interest, what are the use cases for having both auto read and auto write on the same tag?

jkoplo, 2022-11-09:
Bad architecture design? 😉

Good point though. The only cases I can think of are when the PLC code isn't under your control and you're dealing with something suboptimal. Like an HMI could set a value, but you have an alternative way to set it from your code on a different message.

Probably a bit contrived...

kyle-github, 2022-11-09:
I used this automatic read/write to set up a tag wrapper in C++ that would look like the native data type but would automatically show what was in the PLC.   I have run into cases where a single tag is used to trigger an action _and_ see the result.   That's an example of what Jody was talking about.

It is vastly safer to have tags be "one direction."

timyhac, 2024-07-27:
Given that this requires a change to Tag<M,T> (which has been deprecated) - I will mark this as wontfix and close.

closed 2024-07-27 by timyhac