libplctag/libplctag.NET #285: Documentation on CIP performance factors

jkoplo, 2022-10-08:


timyhac, 2023-01-25:
@jkoplo - I propose that we just put forward a single "performance" document that relates to everything we know about performance. We close this one, and rename #284 to "Performance Factors", and expand the scope of what we mean when we say "performance" to include a discussion not just on read latency, but on read throughput, network traffic, PLC CPU load, exception handling overhead, etc. 

It seems different people have different problems, but could all be loosely categorised under the word "performance".

timyhac, 2023-05-20:
I'm going to go with that suggestion - I have written a few words on the general performance story so I will start a new branch and add that in.

closed 2023-05-20 by timyhac