libplctag/libplctag.NET #286: Example for NotifyPropertyChanged

jkoplo, 2022-10-08:


timyhac, 2022-10-18:
I think its going to look something like this in the end, where it extends `Tag<M,T>` by adding a ValueChanged event.

```csharp
class NotifyValueChangedTag<M,T> : Tag<M,T>
        where M : IPlcMapper<T>, new()
{
    public event EventHandler<EventArgs> ValueChanged;

    private int previousHash;

    public NotifyValueChangedTag()
    {
        ReadCompleted += NotifyValueChangedTag_ReadCompleted;
    }

    private void NotifyValueChangedTag_ReadCompleted(object sender, TagEventArgs e)
    {
        var currentHash = this.Value.GetHashCode();
        if (currentHash != previousHash)
        {
            ValueChanged?.Invoke(this, EventArgs.Empty);
        }
        previousHash = currentHash;
    }
}
```

It requires that the type `T` has a `GetHashCode` method that is good at detecting changes. For atomic types, the builtin one should be okay, but for Udts they will need to override it.

```csharp
class MyUdt
{
    public int MyDint { get; set; }
    public float MyFloat { get; set; }

    public override int GetHashCode()
    {
        // Use the tuple shortcut to get an okay hash
        return (MyDint,MyFloat).GetHashCode();
    }

}

class MyUdtMapper : PlcMapperBase<MyUdt>
{
    public override int? ElementSize => 8;

    public override MyUdt Decode(Tag tag, int offset)
    {
        return new MyUdt()
        {
            MyDint = tag.GetInt32(offset),
            MyFloat = tag.GetFloat32(offset + 4)
        };
    }

    public override void Encode(Tag tag, int offset, MyUdt value)
    {
        tag.SetInt32(offset, value.MyDint);
        tag.SetFloat32(offset + 4, value.MyFloat);
    }
}
```

and should be able to used like this:
```csharp
var myTag = new NotifyValueChangedTag<DintPlcMapper, int>()
{
    Name = "MyTag",
    Gateway = "127.0.0.1",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    AutoSyncReadInterval = TimeSpan.FromMilliseconds(500)
};

myTag.ValueChanged += (s, e) => Console.WriteLine($"{myTag.Name} changed at {DateTime.Now} to {myTag.Value}");

var myUdtTag = new NotifyValueChangedTag<MyUdtMapper, MyUdt>()
{
    Name = "MyUdtTag",
    Gateway = "127.0.0.1",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    AutoSyncReadInterval = TimeSpan.FromMilliseconds(500)
};

myUdtTag.ValueChanged += (s, e) => Console.WriteLine($"{myUdtTag.Name} changed at {DateTime.Now} to {myUdtTag.Value.MyDint}|{myUdtTag.Value.MyFloat}");
```

timyhac, 2022-11-20:
`ExampleNotifyChanged` has been added to the examples list.

closed 2022-11-20 by timyhac