libplctag/libplctag.NET #289: Add license notice in file headers

timyhac, 2022-10-12:
When using MPLv2 license, there are recommendations to [add file headers that explain the license.](https://www.mozilla.org/en-US/MPL/2.0/FAQ/#:~:text=add%20the%20header%20from%20Exhibit%20A%20of%20the%20license%20to%20each%20source%20code%20file%20in%20your%20project.)

This was also requested in https://github.com/libplctag/libplctag.NET/issues/248

timyhac, 2022-10-15:
What do we want to put in our license notice?

The MPL FAQ suggests that we add file headers with the following text:
```
This Source Code Form is subject to the terms of the Mozilla Public
License, v. 2.0. If a copy of the MPL was not distributed with this
file, You can obtain one at https://mozilla.org/MPL/2.0/.
```

We can also add a copyright clause. This is not required by MPL but may help to clarify things for library users. Something like:
`Copyright (c) libplctag.NET contributors`
`Copyright (c) 2022 libplctag.NET contributors`
`Copyright (c) libplctag.NET contributors. All Rights Reserved.`

I'm ok to proceed with this file header:
```
// Copyright (c) libplctag.NET contributors
// https://github.com/libplctag/libplctag.NET
//
// This Source Code Form is subject to the terms of the Mozilla Public
// License, v. 2.0. If a copy of the MPL was not distributed with this
// file, You can obtain one at https://mozilla.org/MPL/2.0/.
```

jkoplo, 2022-10-15:
That looks fine to me. Seems excessive to have in every file, but that's probably simplest and unambiguous.

timyhac, 2022-10-15:
I agree about it being excessive - I guess this is about making it easy for our users to understand license implications.

To make this easy to implement I've added an editorconfig to our solution so it should just be a matter of running CodeCleanup and it will automatically add it.

closed 2022-10-15 by timyhac