libplctag/libplctag.NET #290: The process cannot access the file 'plctag.dll'

avnet78, 2022-10-13:
I am using the libplctag.net library. The source code build is deployed on windows 11 server. I’m using dotnet framework version 452.
I have two windows  services or deamon processes.  One reads values from the ethernet IP device and the other one writes values to the ethernet IP device.  Both processes are running simultaneously.
I don’t have any issues with the reads but the writes are throwing exception saying that the DLL is being used by another process. 
Please advise.

timyhac, 2022-10-13:
Set plctag.ForceExtractLibrary = false before any other calls to libplctag

timyhac, 2022-10-14:
@avnet78 - did this solve your problem?

dresslerc, 2022-10-20:
I had same issue.  Did the plctag.ForceExtractLibrary = false and it worked.

Just side note, had to bring in:

`using libplctag.NativeImport;`

avnet78, 2022-10-21:
Yes, it worked for me.

timyhac, 2022-10-21:
Thanks for the update

closed 2022-10-21 by timyhac