libplctag/libplctag.NET #291: Bool tag write not working

avnet78, 2022-10-14:
Here's my sample code. The bool tag write doesn't seem to be working:
I am getting false in both reads.

````
var tag1 = new Tag<BoolPlcMapper, bool>()
  {
      Name = "Glob_RCE_EIP_Bool_2",
      Gateway = deviceIPAddress,
      PlcType = PlcType.ControlLogix,
      Path = tagPath,
      Protocol = Protocol.ab_eip
  };

Console.WriteLine("Writing bool tag");
  tag1.Value = true;
  tag1.Write();
  var tag1Read = tag1.Read();
  Console.WriteLine($"Reading bool tag: success - {tag1Read}");

  Console.WriteLine("Writing bool tag");
  tag1.Value = false;
  tag1.Write();
  tag1Read = tag1.Read();
  Console.WriteLine($"Reading bool tag: success - {tag1Read}");
````

timyhac, 2022-10-14:
We do have outstanding issues with bool mapping because of the myriad of ways boolean values can be represented in PLCs.
Have you read through this ticket? https://github.com/libplctag/libplctag.NET/issues/178

In the example you've posted, the issue could be either read or write.

avnet78, 2022-10-19:
I understand, do you have timeline in mind on when the bool read/write issue will be resolved?
We have tested all other data types and they seems be be pretty accurate. Only issue is with the bool tags.

timyhac, 2022-10-20:
There is no timeline for any development, but you may be able to use the existing bool mapper classes for some of your use cases. If you're writing software for specific hardware models (as opposed to library code which needs to abstract the details of data access) I would recommend using the non-generic Tag class, whose interface is very close to the core library's.

avnet78, 2022-10-26:
Thank you for the update @timyhac

closed 2022-10-26 by avnet78