libplctag/libplctag.NET #301: "ErrorTimeout" exception

avnet78, 2022-10-26:
Below is my sample code. I am trying to read some Float tags from PLC but I am getting "ErrorTimeout" exception. Please help.

````
    internal class Program
    {
        static string tags = "SP_RigWatch_Interface.Read_FLOAT[82],SP_RigWatch_Interface.Read_FLOAT[78],SP_RigWatch_Interface.Read_FLOAT[45],SP_RigWatch_Interface.Read_FLOAT[33],SP_RigWatch_Interface.Read_FLOAT[74],SP_RigWatch_Interface.Read_FLOAT[21],SP_RigWatch_Interface.Read_FLOAT[43],SP_RigWatch_Interface.Read_FLOAT[29],SP_RigWatch_Interface.Read_FLOAT[70],SP_RigWatch_Interface.Read_FLOAT[19],SP_RigWatch_Interface.Read_FLOAT[54],SP_RigWatch_Interface.Read_FLOAT[22],SP_RigWatch_Interface.Read_FLOAT[55],SP_RigWatch_Interface.Read_FLOAT[15],SP_RigWatch_Interface.Read_FLOAT[57],SP_RigWatch_Interface.Read_FLOAT[56],SP_RigWatch_Interface.Read_FLOAT[58],SP_RigWatch_Interface.Read_FLOAT[34],SP_RigWatch_Interface.Read_FLOAT[68],SP_RigWatch_Interface.Read_FLOAT[20],SP_RigWatch_Interface.Read_FLOAT[59],SP_RigWatch_Interface.Read_FLOAT[63],SP_RigWatch_Interface.Read_FLOAT[23],SP_RigWatch_Interface.Read_FLOAT[32],SP_RigWatch_Interface.Read_FLOAT[39],SP_RigWatch_Interface.Read_FLOAT[7],SP_RigWatch_Interface.Read_FLOAT[49],SP_RigWatch_Interface.Read_FLOAT[11],SP_RigWatch_Interface.Read_FLOAT[44],SP_RigWatch_Interface.Read_FLOAT[64],SP_RigWatch_Interface.Read_FLOAT[62]";

        static void Main(string[] args)
        {
            libplctag.NativeImport.plctag.ForceExtractLibrary = false;
            var deviceIPAddress = "192.168.1.195";
            var tagPath = "1,0";
            var eipTags = new List<ITag>();

            try
            {

                Console.WriteLine($"Initializing PLC tags");
                foreach (var item in tags.Split(',').ToList())
                {
                    var plcTag = new Tag<RealPlcMapper, float>()
                    {
                        Name = item,
                        Gateway = deviceIPAddress,
                        Path = tagPath,
                        PlcType = PlcType.ControlLogix,
                        Protocol = Protocol.ab_eip,
                        Timeout = TimeSpan.FromMilliseconds(500)
                    };
                    plcTag.Initialize();
                    eipTags.Add(plcTag);
                }
                Console.WriteLine($"Completed Initializing PLC tags");

                Console.WriteLine($"Reading PLC tags");
                foreach (var item in eipTags)
                {
                    var tagName = item.Name;
                    try
                    {
                        Console.WriteLine($"Reading tag {tagName}");
                        var tagValue = item.Read();
                        Console.WriteLine($"Read result: {tagValue}");
                    }
                    catch (Exception ex)
                    {
                        Console.WriteLine($"Exception faced in reading tag: {tagName}, Error: {ex.Message}, Stacktace: {ex.StackTrace}");
                    }
                }
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Exception faced in the Main method, Error: {ex.Message}, Stacktace: {ex.StackTrace}");
            }

            Console.ReadLine();
        }
    }

````

[image]

ShadowMarker789, 2022-10-27:
Do you still get a timeout if you set the timeout to `TimeSpan.FromSeconds(30)`? Just want to ensure you're not getting real network timeouts. 

I noticed that you have `libplctag.NativeImport.plctag.ForceExtractLibrary = false;` 

If you have upgraded your libplctag.Net version recently, you should ensure that your native `plctag.dll` is matching correctly. I also had weird timeouts when I upgraded libplctag.Net and was using an older version of `plctag.dll`. These issues went away when I used the newest `libplctag.dll`. We have ForceExtractLibrary as false because the application will be expected to run in Program Files without elevated access, so any attempts to write or overwrite those dlls will fail.

avnet78, 2022-10-27:
We have a C language library which is able to read about 100+ tags in milliseconds on the same server.

We want to replace that with a .Net reader. So, the timeout error is a bit misleading.

Thanks,
Vishal

ShadowMarker789, 2022-10-27:
> We have a C language library which is able to read about 100+ tags in milliseconds on the same server. We want to replace that with a .Net reader. So, the timeout error is a bit misleading. Thanks, Vishal

Did changing your version of `plctag.dll` resolve things for you?

timyhac, 2022-11-20:
Hi @avnet78 - if you weren't able to solve this with @ShadowMarker789 suggestion, could you please help to produce a minimal example for further investigation.

Something like this would be helpful:

```csharp
var tag = new Tag<RealPlcMapper, float>()
{
    Name = "SP_RigWatch_Interface.Read_FLOAT[82]",
    Gateway = "192.168.1.195",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(10)
};
tag.Initialize();    // produces LibPlcTagException with status ErrorTimeout immediately after calling.
```

What is the type of Exception you're getting. i.e. is it a LibPlcTagException, or something else?

It would also be helpful if you can attach some debug logs so we can see what is happening inside libplctag. Please study the example here to see how this can be done: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/LoggingExample.cs

closed 2022-12-14 by timyhac