libplctag/libplctag.NET #302: C# Reading string from Micro870

kirillivan0ff, 2022-10-31:
When I'm trying to read a string value from Micro870 I'm receiving libplctag.LibPlcTagException: 'ErrorOutOfBounds'  on Initialize which seems to be a wrong data type? Anyone have some suggestions?
PLC is UP and I'm reading and writing other types of values with no issues (still have a problem with WORD so if someone have examples for it will be happy to see them too)

`        public static string ReadSingleStringTag(formTag readTag)
        {
            var myTag = new Tag<StringPlcMapper, string>()
            {
                Name = readTag.Name,
                Gateway = readTag.Ip,
                PlcType = PlcType.Micro800,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(5)
            };
            myTag.Initialize(); //<-------------- libplctag.LibPlcTagException: 'ErrorOutOfBounds'
            myTag.Read();
            return myTag.Value;
        }`

timyhac, 2022-10-31:
You might need to get into the weeds and figure out how these devices encode strings, libplctag supports extracting different encodings: https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling

The only way to use this from libplctag.NET currently is to use the `Tag` class (not `Tag<M,T>`) as this gives access with minimal wrapping to the underlying libplctag library.

Same with WORD, it depends how you're using it but I'm guessing its a bit string - maybe you're better off using the https://github.com/libplctag/libplctag/wiki/API#bit-accessors.

kyle-github, 2022-11-01:
WORD is 16-bits on most AB PLCs, but I am not sure about Micro800 series.  How many bytes is the result when it is read?

I think someone recently posted that the Micro800 series used strings that had one count byte and then up to 255 bytes of string data.   See the [wiki page about strings](https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling) to see what [attributes](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#optional-generic-string-attributes) to set.

kirillivan0ff, 2022-11-03:
OK, I need to read your links. Basically what you saying it should be like: 
            var myTag = new Tag()
            {
                Name = readTag.Name,
                Gateway = readTag.Ip,
                PlcType = PlcType.Micro800,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(5)
            };
            myTag.Initialize();
            //Transfer from PLC to Value
            myTag.Read();

this way there no errors but I nether don't see any data.

kyle-github, 2022-11-04:
I am not sure what you mean by "no data"?  If there is no error, then you are getting something back from the PLC.   Can you enable debugging mode (level 4) and capture the output so that we can see the packets that go between the PC and PLC?   That may help us understand what is being transmitted.

kirillivan0ff, 2022-11-04:
Hey Kyle, thanks for your response. I'm pretty new to C# so I might need a bit more explanations on this. I assume debug level is set to right value now?
`
            var myTag = new Tag()
            {
                Name = readTag.Name,
                Gateway = readTag.Ip,
                PlcType = PlcType.Micro800,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromSeconds(5),
                DebugLevel = DebugLevel.Spew
            };
            myTag.Initialize();
            //Transfer from PLC to Value
            myTag.Read();
`
The output window at Visual Studio doesn't write anything while reading tags so I probably looking at wrong place or wrong way. It's Windows Forms application if it matters.
BTW should I replace this discussion to Google libplctag group or it's OK to keep it here?

timyhac, 2022-11-04:
@kirillivan0ff - in order to use the debug logging please follow the example here: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/LoggingExample.cs

`myTag.Read()` just brings the data from the PLC into the tag object, you will need to extract that data by using the `myTag.GetString(...)` method.

Given that the StringPlcMapper didnt work, the data extracted from GetString won't work out of the box and you will also need to configure the tag with the string handling parameters. The documentation for that portion of the API is detailed here: https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling. The names are slightly different but you should be able to use Intellisense to discover them.

For example:

```csharp
var myTag = new Tag()
{
    ... Name/Gateway/Protocol etc...
    StringCountWordBytes = 4,
    StringIsByteSwapped = true,
    StringIsCounted = true,
};
```

kirillivan0ff, 2022-11-06:
@timyhac I assume it worked? Please see the log.

`Info            2022-11-06 09:02:25.854 thread(1) tag(0) INFO initialize_modules:177 Starting.
Info            2022-11-06 09:02:25.859 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
Detail          2022-11-06 09:02:25.859 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-06 09:02:25.859 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-06 09:02:25.859 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
Info            2022-11-06 09:02:25.859 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
Info            2022-11-06 09:02:25.859 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
Info            2022-11-06 09:02:25.859 thread(1) tag(0) INFO hashtable_create:75 Starting
Info            2022-11-06 09:02:25.860 thread(1) tag(0) INFO hashtable_create:99 Done
Info            2022-11-06 09:02:25.860 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
Detail          2022-11-06 09:02:25.860 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-06 09:02:25.860 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-06 09:02:25.860 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
Detail          2022-11-06 09:02:25.860 thread(1) tag(0) DETAIL cond_create:1087 Starting.
Detail          2022-11-06 09:02:25.860 thread(1) tag(0) DETAIL cond_create:1114 Done.
Info            2022-11-06 09:02:25.860 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
Detail          2022-11-06 09:02:25.860 thread(1) tag(0) DETAIL thread_create:884 Starting.
Detail          2022-11-06 09:02:25.860 thread(1) tag(0) DETAIL thread_create:917 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO lib_init:161 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
Info            2022-11-06 09:02:25.862 thread(2) tag(0) INFO tag_tickler_func:508 Starting.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO mb_init:2410 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mb_init:2412 Setting up mutex.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO mb_init:2421 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO initialize_modules:231 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO ab_tag_create:173 Starting.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:180
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO rc_alloc_impl:130 Done
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 000001C87E31A8F0
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL ab_tag_create:186 tag=000001C87E31A8F0
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO plc_tag_generic_init_tag:462 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:753 Done.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL mutex_create:753 Done.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL cond_create:1087 Starting.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL cond_create:1114 Done.
Info            2022-11-06 09:02:25.862 thread(1) tag(0) INFO plc_tag_generic_init_tag:493 Done.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL get_plc_type:907 Found Micro8xx PLC.
Detail          2022-11-06 09:02:25.862 thread(1) tag(0) DETAIL ab_tag_create:246 Micro800 needs connected messaging.
Detail          2022-11-06 09:02:25.863 thread(1) tag(0) DETAIL get_plc_type:907 Found Micro8xx PLC.
Detail          2022-11-06 09:02:25.863 thread(1) tag(0) DETAIL session_find_or_create:235 Starting
Detail          2022-11-06 09:02:25.863 thread(1) tag(0) DETAIL session_find_or_create:259 Creating new session.
Info            2022-11-06 09:02:25.863 thread(1) tag(0) INFO session_create_unsafe:451 Starting
Detail          2022-11-06 09:02:25.863 thread(1) tag(0) DETAIL session_create_unsafe:454 Session should use connected messaging.
Info            2022-11-06 09:02:25.863 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:459
[... cut ...]
Info            2022-11-06 09:02:26.255 thread(1) tag(0) INFO hashtable_destroy:272 Done
Info            2022-11-06 09:02:26.258 thread(1) tag(0) INFO lib_teardown:207 Done.
Detail          2022-11-06 09:02:26.263 thread(1) tag(0) DETAIL mutex_destroy:841 destroying mutex 00007FF83CAC6488
Detail          2022-11-06 09:02:26.264 thread(1) tag(0) DETAIL mutex_destroy:854 Done.
Detail          2022-11-06 09:02:26.264 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1303 Starting
2022-11-06 09:02:26.264 thread(1) tag(0) DETAIL plc_tag_unregister_logger:1307 Done.
`

kyle-github, 2022-11-12:
Oops, sorry, let this one slip through the cracks.   It looks like Micro8x0 systems use specially counted strings.  There is a one byte count word followed by the string data bytes without any zero termination.

To copy Tim's example:

```csharp
var myTag = new Tag()
{
    ... Name/Gateway/Protocol etc...
    StringCountWordBytes = 1, // <- this changed
    StringIsByteSwapped = false, // <- this changed
    StringIsCounted = true,
};
```

That should result in the core C DLL figuring out the strings.  Note that writing these could be tricky.   You almost certainly need to set the size of the tag first and then try to set the string value and then write it.

kyle-github, 2022-11-12:
Anyone know how to get a cheap Micro8x0?  There are a lot of questions about them lately.

kirillivan0ff, 2022-11-14:
@kyle-github could you please explain in more details how to set the size of the tag? 
I don't know where to get a cheap one but I'll ping you if I will see something available. If you need to make a tests I can try to give you VPN access to existing one for some time period.

kyle-github, 2022-11-14:
Hi @kirillivan0ff, we are also trying this in another [ticket](https://github.com/libplctag/libplctag/issues/380) for the core DLL.  So far it seems like we are close, but it is not quite working yet.

timyhac, 2023-05-20:
Given that this is being tracked on https://github.com/libplctag/libplctag/issues/380 I will close this issue.

closed 2023-05-20 by timyhac