libplctag/libplctag.NET #303: micro800 does not support Path parameter and providing it throws a timeout exception on Initiasize()

jubeor-gp, 2022-11-09:
**Version:** libplctag 1.1.0

**Problem:** Creating a tag for PLC type micro800 specifying a path will result on a timeout in initialization:

```
Tag aTag = new Tag()
{
    Name = "TagName",
    //Gateway is the IP Address of the PLC or communication module.
    Gateway = "192.168.0.1",
    //Path is the location in the control plane of the CPU. Almost always "1,0".
    Path = _path.Text, // Debugging shows: WARN ab_tag_create:384 A path is not supported for this PLC type.
    PlcType = PlcType.Micro800,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};
aTag.Initialize(); // --> Throws a timeout exception
return aTag;
```

**Work around**: Removing the Path parameters makes things work as expected so far.

I wonder if throwing an exception including the warning message "A path is not supported for this PLC type" or ignoring the parameter would be a good way to help new users of the library.

kyle-github, 2022-11-10:
It isn't quite that simple.   Suppose you have a ControlLogix with two network modules where  one is connected to network A and one to network B.  Suppose you have a Micro800 on network B and a PC on network A.   Using the path parameter and what I call bridging, I can write a path that goes through the ControlLogix from network A to network B and to the Micro800 from the PC.

PC -->  ENBT-A -> ControlLogix backplane -> ENBT-B --> Micro800

In this case (which is not that rare in larger sites) there would be a path parameter, but you would still have a Micro800 at the end.

Here is an example with tag_rw2 to bridge between two ControlLogix systems:

```text
tag_rw2 --type=sint32 '--tag=protocol=ab_eip&gateway=10.206.1.39&path=1,6,18,10.206.1.40,1,4&plc=lgx&name=TestBigArray[0]'
```
In this case, the PLC on the second network happens to be a ControlLogix as well, so there is a path segment for it (`1,4` at the end).  These PLCs are on the same network, but bridging is still used.

timyhac, 2022-11-20:
Thanks for the explanation @kyle-github

closed 2022-11-20 by timyhac