libplctag/libplctag.NET #304: Modbus TCP Async Timeout

nc-koko, 2022-11-15:
Hello all,

I am trying to connect to and read from a Modbus TCP PLC tag and I keep getting timeouts when using the async methods. In InitializeAsync, it always comes back with:

`libplctag.LibPlcTagException: 'ErrorTimeout'`

Same happens with ReadAsync. If I instead use the synchronous methods Initialize() and Read(), and leave everything else as is, it works fine. The async methods work fine with the ab_eip protocol, it's just the protocol modbus_tcp where I get the exceptions.

Am I missing something here?

timyhac, 2022-11-17:
Please see this example that shows how to collection libplctag debug logs.
https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/LoggingExample.cs

nc-koko, 2022-11-21:
Here is the log:
```
Info            2022-11-21 11:45:32.909 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO initialize_modules:177 Starting.
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
Detail          2022-11-21 11:45:32.920 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-21 11:45:32.920 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO hashtable_create:75 Starting
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO hashtable_create:99 Done
Info            2022-11-21 11:45:32.920 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
Detail          2022-11-21 11:45:32.920 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-21 11:45:32.920 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL cond_create:1087 Starting.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL cond_create:1114 Done.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL thread_create:884 Starting.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL thread_create:917 Done.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO lib_init:161 Done.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
Info            2022-11-21 11:45:32.921 thread(2) tag(0) INFO tag_tickler_func:508 Starting.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
Info            2022-11-21 11:45:32.921 thread(1) tag(0) INFO mb_init:2410 Starting.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL mb_init:2412 Setting up mutex.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-21 11:45:32.921 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-21 11:45:32.923 thread(1) tag(0) INFO mb_init:2421 Done.
Info            2022-11-21 11:45:32.924 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
Info            2022-11-21 11:45:32.924 thread(1) tag(0) INFO initialize_modules:231 Done.
Info            2022-11-21 11:45:32.924 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=modbus_tcp
Info            2022-11-21 11:45:32.924 thread(1) tag(0) INFO mb_tag_create:276 Starting.
Info            2022-11-21 11:45:32.925 thread(1) tag(0) INFO create_tag_object:332 Starting.
Info            2022-11-21 11:45:32.925 thread(1) tag(0) INFO parse_register_name:1983 Starting.
Detail          2022-11-21 11:45:32.925 thread(1) tag(0) DETAIL parse_register_name:2007 Found holding register type.
Info            2022-11-21 11:45:32.925 thread(1) tag(0) INFO parse_register_name:2019 Done.
Detail          2022-11-21 11:45:32.925 thread(1) tag(0) DETAIL create_tag_object:367 Tag data size is 2 bytes.
Info            2022-11-21 11:45:32.926 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from create_tag_object:370
Info            2022-11-21 11:45:32.926 thread(1) tag(0) INFO rc_alloc_impl:130 Done
Detail          2022-11-21 11:45:32.926 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 08557B18
Info            2022-11-21 11:45:32.926 thread(1) tag(0) INFO create_tag_object:401 Done.
Info            2022-11-21 11:45:32.926 thread(1) tag(0) INFO plc_tag_generic_init_tag:462 Starting.
Detail          2022-11-21 11:45:32.927 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-21 11:45:32.927 thread(1) tag(0) DETAIL mutex_create:753 Done.
Detail          2022-11-21 11:45:32.927 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-21 11:45:32.927 thread(1) tag(0) DETAIL mutex_create:753 Done.
Detail          2022-11-21 11:45:32.927 thread(1) tag(0) DETAIL cond_create:1087 Starting.
Detail          2022-11-21 11:45:32.929 thread(1) tag(0) DETAIL cond_create:1114 Done.
Info            2022-11-21 11:45:32.929 thread(1) tag(0) INFO plc_tag_generic_init_tag:493 Done.
Info            2022-11-21 11:45:32.930 thread(1) tag(0) INFO find_or_create_plc:475 Starting.
Detail          2022-11-21 11:45:32.930 thread(1) tag(0) DETAIL find_or_create_plc:509 Creating new PLC connection.
Info            2022-11-21 11:45:32.930 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from find_or_create_plc:513
[... cut ...]
   at libplctag.NativeTagWrapper.InitializeAsync(CancellationToken token) in C:\Users\NK\test5\libplctag.NET\src\libplctag\NativeTagWrapper.cs:line 346
   at libplctag.Tag`2.InitializeAsync(CancellationToken token) in C:\Users\NK\test5\libplctag.NET\src\libplctag\TagOfT.cs:line 155
   at CSharpDotNetCore.ExampleAsync.Run() in C:\Users\NK\test5\libplctag.NET\src\Examples\CSharp DotNetCore\ExampleAsync.cs:line 37
   at CSharpDotNetCore.Program.Main(String[] args) in C:\Users\NK\test5\libplctag.NET\src\Examples\CSharp DotNetCore\Program.cs:line 31
   at CSharpDotNetCore.Program.<Main>(String[] args)
```

kyle-github, 2022-11-21:
I can see the TCP connection get set up.  Then I see the initial read.  That works.  Then there's nothing at all from outside the library.  I don't see any attempts to start another read.

I don't see any logging about tag created events or read events completing.   I wonder if there is a problem in the core library here.  Let me do some investigation to make sure all the events are getting sent correctly.   

Also, it looks like the new(ish) tag create function is not being used, so there isn't a callback during the early part of the tag and PLC connection set up.   It is possible that the event is being missed due to that.

timyhac, 2022-11-22:
> Also, it looks like the new(ish) tag create function is not being used, so there isn't a callback during the early part of the tag and PLC connection set up. It is possible that the event is being missed due to that.

I'm not sure how I missed this, this is probably it - libplctag.NET does not use `plc_tag_create_ex` yet.

timyhac, 2022-11-23:
@nc-koko - there is some work on this front in [this branch](https://github.com/libplctag/libplctag.NET/pull/307).

@jkoplo - ok if we release this as 1.1.1-alpha?

jkoplo, 2022-11-24:
Do you want me to do a quick review in that branch before we cut a release?

But yeah, if there's a new method we missed we should probably update the pre-release packages asap.

timyhac, 2022-11-24:
Hi @nc-koko - an alpha package v1.1.1-alpha.0 has been released with a fix - could you please test it out?

nc-koko, 2022-11-25:
I'm still seeing the same exception. Here's the debug log

```
Info            2022-11-25 11:51:19.764 thread(1) tag(0) INFO plc_tag_create_ex:833 Starting
Info            2022-11-25 11:51:19.776 thread(1) tag(0) INFO initialize_modules:177 Starting.
Info            2022-11-25 11:51:19.776 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
Detail          2022-11-25 11:51:19.776 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-25 11:51:19.776 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-25 11:51:19.776 thread(1) tag(0) INFO initialize_modules:207 Initialized library modules.
Info            2022-11-25 11:51:19.777 thread(1) tag(0) INFO lib_init:135 Setting up global library data.
Info            2022-11-25 11:51:19.777 thread(1) tag(0) INFO lib_init:137 Creating tag hashtable.
Info            2022-11-25 11:51:19.777 thread(1) tag(0) INFO hashtable_create:75 Starting
Info            2022-11-25 11:51:19.777 thread(1) tag(0) INFO hashtable_create:99 Done
Info            2022-11-25 11:51:19.777 thread(1) tag(0) INFO lib_init:143 Creating tag hashtable mutex.
Detail          2022-11-25 11:51:19.777 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-25 11:51:19.777 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-25 11:51:19.777 thread(1) tag(0) INFO lib_init:149 Creating tag condition variable.
Detail          2022-11-25 11:51:19.777 thread(1) tag(0) DETAIL cond_create:1087 Starting.
Detail          2022-11-25 11:51:19.778 thread(1) tag(0) DETAIL cond_create:1114 Done.
Info            2022-11-25 11:51:19.778 thread(1) tag(0) INFO lib_init:155 Creating tag tickler thread.
Detail          2022-11-25 11:51:19.778 thread(1) tag(0) DETAIL thread_create:884 Starting.
Detail          2022-11-25 11:51:19.778 thread(1) tag(0) DETAIL thread_create:917 Done.
Info            2022-11-25 11:51:19.778 thread(1) tag(0) INFO lib_init:161 Done.
Info            2022-11-25 11:51:19.778 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
Info            2022-11-25 11:51:19.778 thread(2) tag(0) INFO tag_tickler_func:508 Starting.
Info            2022-11-25 11:51:19.779 thread(1) tag(0) INFO ab_init:123 Initializing AB protocol library.
Detail          2022-11-25 11:51:19.779 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-25 11:51:19.779 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-25 11:51:19.779 thread(1) tag(0) INFO ab_init:132 Finished initializing AB protocol library.
Info            2022-11-25 11:51:19.779 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
Info            2022-11-25 11:51:19.780 thread(1) tag(0) INFO mb_init:2410 Starting.
Detail          2022-11-25 11:51:19.780 thread(1) tag(0) DETAIL mb_init:2412 Setting up mutex.
Detail          2022-11-25 11:51:19.780 thread(1) tag(0) DETAIL mutex_create:726 Starting.
Detail          2022-11-25 11:51:19.780 thread(1) tag(0) DETAIL mutex_create:753 Done.
Info            2022-11-25 11:51:19.780 thread(1) tag(0) INFO mb_init:2421 Done.
Info            2022-11-25 11:51:19.780 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
Info            2022-11-25 11:51:19.781 thread(1) tag(0) INFO initialize_modules:231 Done.
Detail          2022-11-25 11:51:19.783 thread(1) tag(0) DETAIL attr_create_from_str:131 Starting.
Detail          2022-11-25 11:51:19.783 thread(1) tag(0) DETAIL attr_create_from_str:160 Key-value pair "protocol=modbus_tcp".
Detail          2022-11-25 11:51:19.784 thread(1) tag(0) DETAIL attr_create_from_str:175 Key-value pair before trimming "protocol":"modbus_tcp".
Detail          2022-11-25 11:51:19.785 thread(1) tag(0) DETAIL attr_create_from_str:187 Key-value pair after trimming "protocol":"modbus_tcp".
Detail          2022-11-25 11:51:19.785 thread(1) tag(0) DETAIL attr_create_from_str:160 Key-value pair "gateway=10.106.2.20".
Detail          2022-11-25 11:51:19.786 thread(1) tag(0) DETAIL attr_create_from_str:175 Key-value pair before trimming "gateway":"10.106.2.20".
Detail          2022-11-25 11:51:19.786 thread(1) tag(0) DETAIL attr_create_from_str:187 Key-value pair after trimming "gateway":"10.106.2.20".
Detail          2022-11-25 11:51:19.787 thread(1) tag(0) DETAIL attr_create_from_str:160 Key-value pair "path=0".
Detail          2022-11-25 11:51:19.789 thread(1) tag(0) DETAIL attr_create_from_str:175 Key-value pair before trimming "path":"0".
Detail          2022-11-25 11:51:19.789 thread(1) tag(0) DETAIL attr_create_from_str:187 Key-value pair after trimming "path":"0".
Detail          2022-11-25 11:51:19.790 thread(1) tag(0) DETAIL attr_create_from_str:160 Key-value pair "plc=controllogix".
Detail          2022-11-25 11:51:19.790 thread(1) tag(0) DETAIL attr_create_from_str:175 Key-value pair before trimming "plc":"controllogix".
Detail          2022-11-25 11:51:19.791 thread(1) tag(0) DETAIL attr_create_from_str:187 Key-value pair after trimming "plc":"controllogix".
Detail          2022-11-25 11:51:19.793 thread(1) tag(0) DETAIL attr_create_from_str:160 Key-value pair "elem_size=4".
Detail          2022-11-25 11:51:19.793 thread(1) tag(0) DETAIL attr_create_from_str:175 Key-value pair before trimming "elem_size":"4".
Detail          2022-11-25 11:51:19.794 thread(1) tag(0) DETAIL attr_create_from_str:187 Key-value pair after trimming "elem_size":"4".
[... cut ...]
Detail          2022-11-25 11:51:20.767 thread(3) tag(0) DETAIL tickle_all_tags:1077 Done: PLCTAG_STATUS_OK
Detail          2022-11-25 11:51:20.768 thread(3) tag(0) DETAIL modbus_plc_handler:762 in PLC_READY state.
Detail          2022-11-25 11:51:20.768 thread(3) tag(0) DETAIL socket_wait_event:1612 Starting.
Info            2022-11-25 11:51:20.768 thread(4) tag(11) INFO plc_tag_unregister_callback:1226 Starting.
Info            2022-11-25 11:51:20.769 thread(4) tag(11) INFO plc_tag_unregister_callback:1245 Done.

```

kyle-github, 2022-11-25:
I don't see any calls to the event dispatch code in this.   I do see a few calls to plc_tag_status() but that is only during the tail end of tag creation.   It looks like the tag is being created without any timeout on plc_tag_create_ex()?

Let me do some experimentation to see if I can replicate this locally with just the C DLL.

timyhac, 2022-11-25:
> It looks like the tag is being created without any timeout on plc_tag_create_ex()

That is correct, for Async initialisation we pass in a timeout of 0 and manage the timeout separately.

timyhac, 2022-11-25:
After doing some more testing I found an issue relating to synchonous `Initialize` and have published a new alpha, I don't think that will affect this issue however.

kyle-github, 2022-11-26:
This looks a bit odd.   The only events I see are aborts.   And those are triggered from user code.  I do not see any read completion events or creation completion events.

I changed one of the examples to use Modbus and I saw all the events I expected, so they are being generated in the C DLL.

If I had to guess what is happening, I would guess that for some reason the creation completed event is either being missed or there isn't a valid callback when it happens.   After that, the user code is just waiting.  There are occasional calls to plc_tag_abort() that seem to happen after long periods of time.  

If the PLCTAG_EVENT_CREATED event is being missed, then the user code will probably not try to read.   There is a read completing, but it is the internally triggered initial read that happens just after the tag is created.  I do not even see those events.

The fact that I see some but not others is confusing.

timyhac, 2022-11-27:
If someone could point me in the right direction of setting up a local debugging environment I can do a bit more.

I've been using this container as a modbus server: https://hub.docker.com/r/oitc/modbus-server
And using this code. It is C# but directly uses the native methods.

```csharp
var tagHandle = libplctag.NativeImport.plctag.plc_tag_create("protocol=modbus_tcp&gateway=127.0.0.1:5020&path=0&plc=controllogix&elem_size=4&name=hr0", 1000);
var readStatus = (Status)libplctag.NativeImport.plctag.plc_tag_read(tagHandle, 1000);
var value = libplctag.NativeImport.plctag.plc_tag_get_int32(tagHandle, 0);
var status = (Status)libplctag.NativeImport.plctag.plc_tag_status(tagHandle); // ErrorOutOfBounds
```

kyle-github, 2022-11-28:
That looks pretty good.   You may want to check the IP address.  Is that what your container will run on? 

The plc attribute is completely unused.   The element size is going to be overridden and will be 2 (Modbus registers are 16 bits).

The default is to read only one element which in this case will be one register, so only 2 bytes.   The `plc_tag_get_int32()` call will fail because there isn't enough data in the tag buffer.  You'll get back -2 billion IIRC (INT32_MIN in C).  Use `plc_tag_get_int16()` instead.

Try this as a tag attribute string: `protocol=modbus_tcp&gateway=127.0.0.1:5020&path=0&elem_size=2&elem_count=1&name=hr1`.

Also Modbus has weird start-from-one counting in some cases.   Different devices have different defaults.   So `hr0` might be illegal for that server.  I switched it to `hr1` above just to be sure. Based on the comments on the Docker site, I am guessing that this one starts with zero.

timyhac, 2022-11-28:
> The element size is going to be overridden and will be 2 (Modbus registers are 16 bits).
Yep this was the solution, using the `plc_tag_get_int16()` method works.

I've done some testing and to me it looks like we aren't getting the Created, ReadCompleted and WriteCompleted events when timeout is 0.

The test creates an int16 tag, writes a value to it, and reads it back.

When using a timeout of 1000, we get all 5 events, and it succesfully writes and then reads the value.
```
11    | Created              | Ok    | 0
11    | WriteStarted         | Ok    | 0
11    | WriteCompleted       | Ok    | 0
11    | ReadStarted          | Ok    | 0
11    | ReadCompleted        | Ok    | 0
```
[timeout1000.log](https://github.com/libplctag/libplctag.NET/files/10107880/timeout1000.log)

When using a timeout of 0, we get these two only
```
11    | WriteStarted         | Ok    | 0
11    | ReadStarted          | Ok    | 0
```
[timeout0.log](https://github.com/libplctag/libplctag.NET/files/10107887/timeout0.log)

The code listing is below:
```csharp
public static void Run()
{
    LibPlcTag.LogEvent += LibPlcTag_LogEvent;
    LibPlcTag.DebugLevel = DebugLevel.Detail;

    var timeout = 1000;
    var attributeString = "protocol=modbus_tcp&gateway=127.0.0.1:5020&path=0&elem_size=2&name=hr0";
    var tagHandle = plctag.plc_tag_create_ex(attributeString, eventCallback, IntPtr.Zero, timeout); // 11

    WaitUntilNotPending(tagHandle);

    var setStatus = (Status)plctag.plc_tag_set_int16(tagHandle, 0, 2345); // Ok
    var writeStatus = (Status)plctag.plc_tag_write(tagHandle, timeout); // Ok

    WaitUntilNotPending(tagHandle);

    var readStatus = (Status)plctag.plc_tag_read(tagHandle, timeout); // Ok

    WaitUntilNotPending(tagHandle);

    var value = plctag.plc_tag_get_int16(tagHandle, 0);  // 2345
    var status = (Status)plctag.plc_tag_status(tagHandle); // Ok
}

private static void WaitUntilNotPending(int tagHandle)
{
    STATUS_CODES status;
    do status = (STATUS_CODES)plctag.plc_tag_status(tagHandle);
    while (status == STATUS_CODES.PLCTAG_STATUS_PENDING);
    if (status != STATUS_CODES.PLCTAG_STATUS_OK) throw new Exception(status.ToString());
}

private static void eventCallback(int tag_id, int event_id, int status, IntPtr userData)
{
    Console.WriteLine($"____LIBPLCTAG_NET____   {tag_id,-5} | {(Event)event_id,-20} | {(Status)status,-5}");
}

private static void LibPlcTag_LogEvent(object sender, LogEventArgs e)
{
    Console.WriteLine(e.Message);
}
```

I used the container without any configuration customization, with whatever the defaults they supply are.
The container app itself is pretty good I think, it runs in a debug mode by default and shows incoming/outgoing data.

kyle-github, 2022-11-29:
That matches the problem that the OP saw.   Let me redo this exact example in C and see if I can repeat it.  I think the example I converted was using synchronous (waiting) calls.

kyle-github, 2022-11-29:
There was definitely a bug here and a hole in my testing.   The created event is not raised when creating a tag in async mode.   See issue [404](https://github.com/libplctag/libplctag/issues/404) in the core C library.

I have added a couple of tests and fixed (I think) the creation event problem.   I have all this in the prerelease branch of the core C library.   

I refactored some of the Modbus code to directly generate events within the protocol specific part of the library.   This generates all the events that it should both in synchronous and asynchronous mode.   

@timyhac can you test the prerelease branch?   I want to make sure that this is working before trying a full release.

timyhac, 2022-11-29:
@kyle-github - it looks like we're getting the events now.

The way libplctag.NET is implemented is that it waits for a created event with Status=Ok.

The prerelease as it stands triggers one callback with Event=Created and Status=Pending, and so never completes the callback.

To get around this we'd need to poll the status until we get Status=Ok - do you think thats the right way to go?

kyle-github, 2022-11-29:
The PLCTAG_EVENT_CREATED event is raised with a status of PLCTAG_STATUS_PENDING when you use async tag creation.  That is the actual status as the tag is not fully ready when the handle is returned.   If there is an abort, then you get a PLCTAG_EVENT_ABORTED with a status of PLCTAG_ERR_ABORT.  You can get non-OK statuses when there is an error for all events.

I will dig into what it would take to return a PLCTAG_EVENT_CREATED once the tag is ready.  That only happens when the initial read is done (but this does not work for list or raw tags since they do not do an initial read).

I think we've discussed this before and I thought we'd found a solution.  It doesn't seem to be working though.

kyle-github, 2022-11-29:
OK.  Another attempt in the `prerelease` branch.   I also updated my tests to check whether there are read events coming in before the create finished event, and fail if one is seen.

You should see a created event with a status of OK now.   If it is actually OK.

timyhac, 2022-11-29:
> I think we've discussed this before and I thought we'd found a solution.

I recall discussing things surrounding this topic but not what the solution was.

The updated prerelease looks to be creating the event with Ok now. Looks good!
I've only tested the happy path, so I'll try to do some further testing over the next few days in various connectedness scenarios.

I think this has been a good exercise as we've been able to find a few problems both in native library and the wrapper.

One issue that I'm getting now is that, in some situations, the modbus teardown doesn't complete.
This only occurs when running a Debug build of the .NET test harness (I'm not sure if the native binary has a debug/release equivalent - I just built the dll using the default CMake configuration).
When running the Release build, it does complete.
I'm not sure how big of an issue this is, but I did note that there are some users that require a call to shutdown, so this may be important for them.

While it almost never completes when running logging at log level = DETAIL or lower, it does complete about 75% of the time when running at LogLevel = SPEW.

[Debug.log](https://github.com/libplctag/libplctag.NET/files/10117647/Debug.log)
[Release.log](https://github.com/libplctag/libplctag.NET/files/10117650/Release.log)

I'm testing on Windows x86-64

kyle-github, 2022-11-29:
Looking at the logs is confusing.  I do not see any calls to `plc_tag_destroy()` or `plc_tag_shutdown()`.  I am not sure why there is any teardown happening unless it is the `atexit()` stuff (you can hook into program shutdown in POSIX systems and Windows sorta-kinda support that, sometimes).  This does bring up that I should hook that to call the new `plc_tag_shutdown()` instead of just the old shutdown code.  The new version of `plc_tag_shutdown()` tries pretty hard to free up all resources.

kyle-github, 2022-11-29:
I tried hooking the POSIX and Windows DLL exit hooks to call `plc_tag_shutdown()`.   It calls it twice (which is fine) when you call it and normally exist after that.  The `prerelease` branch is updated.

This should probably be a new issue.

timyhac, 2022-11-30:
This looks to be resolved now - once we're happy with the core library and that is released, I'll update the wrapper packages.

timyhac, 2022-12-01:
Hi @nc-koko - could you please try testing with [libplctag-v1.1.1-beta.0](https://github.com/libplctag/libplctag.NET/releases/tag/libplctag-v1.1.1-beta.0).

nc-koko, 2022-12-02:
@timyhac Async methods are working for me again! Thanks for the fix.

timyhac, 2022-12-02:
@nc-koko - Thanks for the feedback and for reporting the issue! We'll leave it as beta for a little while and then release as a non-prerelease

kyle-github, 2022-12-02:
Woot!  Thanks for calling this to our attention, @nc-koko!

timyhac, 2022-12-14:
It has been a couple so will close this issue and release as a production version.

closed 2022-12-14 by timyhac