libplctag/libplctag.NET #305: Reading Raw Tags

airwedge1, 2022-11-17:
I see an example of raw tags from the base C Library (https://github.com/libplctag/libplctag/blob/release/src/examples/test_raw_cip.c) , but I can't seem how to figure out how to do the same thing with the .net wrapper.  

Anyone have an example using the .net wrapper?

timyhac, 2022-11-17:
You can't do it with `Tag<T>` yet but you should be able to with the `Tag` class.

Please study this example here: https://github.com/libplctag/libplctag.NET/blob/%23211/src/Examples/CSharp%20DotNetCore/ExampleRaw.cs

Its in a branch for some unrelated feature but should still work with the current build.

What is your use case by the way?

airwedge1, 2022-11-18:
I'm trying to read and write the wall clock time. https://github.com/libplctag/libplctag/issues/360.  Thanks for the example.  I'll check it out.

kyle-github, 2022-11-19:
Ooh, forgot about that one.  I need to triage the open tickets again.   I should code up an example in C for this though I bet that C# will be a lot more fun.

airwedge1, 2022-11-21:
The ExampleRaw.cs example did not work out of the box for me.  It throws an ErrorBadParam exception.  The initialization order does not seem to be correct.

I simplified the example and got a lot farther combining the C example with your example. I get data back but the decode returns an ErrorOutOfBounds exception.  My assumption was the example payload here was the same as the @tags payload.  

`
            var raw_payload = new byte[] {
                0x55,
                0x03,
                0x20,
                0x6b,
                0x25,
                0x00,
                0x00,
                0x00,
                0x04,
                0x00,
                0x02,
                0x00,
                0x07,
                0x00,
                0x08,
                0x00,
                0x01,
                0x00
            };

            var x = new Tag()
            {
                Gateway = "0.0.0.0",
                Path = "1,0",
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Timeout = TimeSpan.FromMilliseconds(1000),
                Name = "@raw"
            };

            x.Initialize();

            x.SetSize(raw_payload.Length);

            for (int ii = 0; ii < raw_payload.Length; ii++)
                x.SetUInt8(ii, raw_payload[ii]);

            x.Write();

            var tagInfoMapper = new TagInfoPlcMapper() { PlcType = PlcType.ControlLogix };
            var info = tagInfoMapper.Decode(x);
`

airwedge1, 2022-11-22:
Correct If I do the below it works fine.  Comparing between @tags and @raw it looks like the very first request is identical, but I think the @tags actually makes multiple requests, so I actually think the @raw example is working correctly. I attached the logs anyway. 

So I just need to figure out what the raw payload needs to look like for reading / writing the wall clock time.  I know there something called a class code = x8b, instance = x01, something called request data that is \x01\x00\x0B\x00 when reading, and is a LINT value when writing.

I will close this issue.  I still have the other ticket about the wall clock time that Kyle mentioned so I'll leave that one open.

```

var x = new Tag()
{
    Gateway = "10.1.16.11",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromMilliseconds(1000),
    Name = "@tags"
};

x.Read();

var tagInfoMapper = new TagInfoPlcMapper() { PlcType = PlcType.ControlLogix };
var info = tagInfoMapper.Decode(x);
```
[logs.zip](https://github.com/libplctag/libplctag.NET/files/10068831/logs.zip)