libplctag/libplctag.NET #311: Read Tags - Bool Vector

VitorKawao, 2022-11-30:
Hello all,
I am trying to read all tags from the plc.
I am using the code below

var tags = new Tag<TagInfoPlcMapper, TagInfo[]>
  {
      Gateway = myIp,
      Path = null,
      PlcType = PlcType.Micro800,
      Protocol = Protocol.ab_eip,
      Name = "@tags"
  };
  tags.Read();

It works fine for all tags except for bool vectors.

[image]

kyle-github, 2022-11-30:
"Works fine" is not enough detail to work with here :-)

Does the tag listing show the bool vector tag at all or is it just missing from the listing?  In what way is the listing incorrect?  Anything you can add will help.

VitorKawao, 2022-12-01:
The vector of bool is not listed.
It shows all other vectors (vector of real, sint, ulint, usint, etc)

kyle-github, 2022-12-01:
Can you use the `list_tags_micro8x0` command line tool that comes with the C DLL to try to dump the tags out?  I see two possibilities:

1. Something is filtering out these kinds of tags,
2. The Micro800 is not sending them back at all.

The program I noted above does no filtering.  It even shows system tags and other internal objects that normally are not visible.

You can download a binary release containing the core C DLL and these programs [here](https://github.com/libplctag/libplctag/releases/tag/v2.5.5).

If the problem is the second one above, then there might not be much we can do without documentation on how these objects are being found by CCW.

VitorKawao, 2022-12-05:
Hello,
I used the list_tags_micro8x0 from the c dll.
It had the same problem, it founds all other tags but it does not found the bool vector

kyle-github, 2022-12-05:
That is not good.  The Micro800 is not returning the vector of bool in the listing at all.   Let me ask around to other library owners to see if anyone has seen this.

kyle-github, 2022-12-13:
I have confirmation from another library in Python (the library is pylogix) that reading BOOL arrays does not work.   Since we use the same logic, this is not a surprise, but it is annoying.  The author of pylogix is trying to get his Micro800 back (he loaned it out) to do more testing.

It looks like this is going to take a while. 

Thanks for bringing this up!

timyhac, 2022-12-14:
Given that this is a hardware/protocol/core library problem - can we track this in the core library?

kyle-github, 2022-12-15:
D'oh.  Issue [408](https://github.com/libplctag/libplctag/issues/408).

timyhac, 2023-05-20:
Given that this is a core issue and is tracked there, I will close this issue.

closed 2023-05-20 by timyhac