libplctag/libplctag.NET #312: Modbus teardown does not complete

kyle-github, 2022-11-30:
(taken from #304 as a separate issue)

@timyhac said:

> One issue that I'm getting now is that, in some situations, the modbus teardown doesn't complete.
> This only occurs when running a Debug build of the .NET test harness (I'm not sure if the native binary has a debug/release equivalent - I just built the dll using the default CMake configuration).
> When running the Release build, it does complete.
> I'm not sure how big of an issue this is, but I did note that there are some users that require a call to shutdown, so this may be important for them.
>
> While it almost never completes when running logging at log level = DETAIL or lower, it does complete about 75% of the time when running at LogLevel = SPEW.

kyle-github, 2022-11-30:
Looking at the logs is confusing. I do not see any calls to plc_tag_destroy() or plc_tag_shutdown(). I am not sure why there is any teardown happening unless it is the atexit() stuff (you can hook into program shutdown in POSIX systems and Windows sorta-kinda support that, sometimes). This does bring up that I should hook that to call the new plc_tag_shutdown() instead of just the old shutdown code. The new version of plc_tag_shutdown() tries pretty hard to free up all resources.

I tried hooking the POSIX and Windows DLL exit hooks to call plc_tag_shutdown(). It calls it twice (which is fine) when you call it and normally exist after that. The prerelease branch is updated.

This should probably be a new issue.

kyle-github, 2022-11-30:
The latest `prerelease` branch of the core library has an attempt at fixing this.   I was able to hook into the exit path.  At least on macOS calling `plc_tag_shutdown()` ends up calling that function twice.   This is supported and is safe.

timyhac, 2022-11-30:
Yep this looks good now, I haven't been able to repeat the problem when trying to with the same steps.

kyle-github, 2022-11-30:
I will try to release the core library today or tomorrow.

closed 2022-12-01 by timyhac