libplctag/libplctag.NET #314: Current async implementation drives too much allocation

jkoplo, 2022-12-09:
The way async is currently implemented creates a Task Completion Source for each tag which is fine for low tag counts, but causes a lot of allocation when trying to read large (>1000) numbers of tags as fast as possible.

After reading some posts by Cleary and Toub, I've realized that the more _correct_ implementation is to not have 1000s of separate awaits simultaneously and instead of trying to await a bunch of tasks it's better to bundle that together into a single task - especially in the case that we *know* we want to read a group of tags and wait for all of them.

This is probably a breaking change - but we'll see once I investigate more.

timyhac, 2022-12-09:
Good stuff! It would be good to have a benchmarks project that we can run to analyse alternative implementations. Benchmarks.NET has a memory diagnoser - I reckon that's the go.

Can you point to the source you're reading, I'm also interested in this area

jkoplo, 2022-12-10:
> Good stuff! It would be good to have a benchmarks project that we can run to analyse alternative implementations. Benchmarks.NET has a memory diagnoser - I reckon that's the go.

I actually merged in a benchmark project last night. It's the one I used on that old performance issue to compare different ways of reading lots of tags. It's not fancy - just stopwatches the reads. That said, when I ran it before merging it in I didn't see nearly the amount of crazy GC I used to and the CPU was lower (I think).

> Can you point to the source you're reading, I'm also interested in this area

Of course I lost the really canonical best practice I found from one of those guys. Here's the articlesI was smart enough to save: 
https://github.com/dotnet/runtime/discussions/45877
https://www.danielcrabtree.com/blog/248/maximizing-throughput-the-overhead-of-1-million-tasks
https://stackoverflow.com/questions/66387225/awaiting-a-single-net-event-with-a-valuetask
https://stackoverflow.com/questions/69147931/avoiding-allocations-and-maintaining-concurrency-when-wrapping-a-callback-based
https://minidump.net/performance-best-practices-in-c-b85a47bdd93a

I actually came across that same Gravell library, but I got there from a git issue somewhere! I think that library might be unnecessary on later dotnet revisions...

>Is this an area where the C DLL could help more? Something like tag groups but implemented in the lower level DLL? I want to brainstorm about possible v3 APIs that would help.

Honestly, I'm not sure yet. From my understanding, all the events/callbacks from the base library use the same thread. So when we ask it to read 1000 tags and then wait for all the tags to be read, we're pretending like all those 1000 tags will have overlapping/independent completion (callback) times. They won't - the callbacks we get will all be one after another just not sequential. We're essentially treating it like hitting 1000 APIs on different servers and waiting for them to all complete, whereas it's really all coming from the same 'server' and the completions are more or less sequential.

The way we're handling async/await makes perfect sense for individual tags. If you had a ton of independent parts of your code and they all just call tags and wait on them whenever - then you don't care that other tags might be getting read too. But if you _know_ you're going to read 1000 tags and wait for all of them to be read and then continue your code then the overhead of 1000 tasks is wasted.

With all that said, I don't think doing a 'group read' in the lower library is necessary and might be over optimizing. I want to test a few things out first.

kyle-github, 2022-12-10:
There is one thread per connection to any given PLC.   Generally, if you have three PLCs, you'll have three threads.   Each thread can call the event callbacks.   If you have 1000 tags all on one PLC, you will get the callbacks sequentially.   If you have those tags spread across 10 PLCs, you'll get get callbacks from different threads.  _BUT_ the callbacks are only called when the tag's mutex is held, so you won't get simultaneous callbacks.   That's the theory at least.

timyhac, 2022-12-14:
@jkoplo - it would be ideal to generate a benchmark solution, so that we can get hard numbers across all of our target platforms. Something like this with an appropriate Mock would be enough I think:

```csharp
using BenchmarkDotNet.Attributes;
using Moq;

namespace libplctag.Benchmarks
{
    [MemoryDiagnoser]
    public class ConcurrentReads
    {
        private const int numberOfTags = 10;

        private List<NativeTagWrapper> tags;

        [GlobalSetup]
        public void Setup()
        {
            var minimumTagId = 11;
            tags = Enumerable.Range(minimumTagId, numberOfTags)
                            .Select(i => (i, GetMock(i)))
                            .Select(x => {
                                var (tagId, mock) = x;
                                var tag = new NativeTagWrapper(mock.Object);
                                tag.Initialize();
                                return tag;
                                })
                            .ToList();
        }

        [Benchmark]
        public async Task Run()
        {
            var reads = tags.Select(tag => tag.ReadAsync()).ToArray();
            await Task.WhenAll(reads);
        }

        public int[] GetValues()
        {
            return tags.Select(tag => tag.GetInt32(0)).ToArray();
        }
    }
}

```

jkoplo, 2022-12-15:
@timyhac - yeah I should measure before I 'improve' -I know better... 

I'll throw something together in the next week or two on a new issue.

timyhac, 2023-05-20:
@jkoplo - is there still some work happening in this space? You mentioned in the previous comment that when you do throw something together it will be in a new issue - hence I will close this one.

closed 2023-05-20 by timyhac