libplctag/libplctag.NET #317: plctag.dll Still in Use After Shutdown

kyle-github, 2022-12-09:
I am opening this to track the [issue in the forum](https://groups.google.com/g/libplctag/c/4-Cis-R6xTo).  Now that `plc_tag_shutdown()` cleans up everything, the DLL should be ready to be unloaded.   That said, I think that there is a sort of reference count that Windows maintains and the ref count for the native DLL is not going to zero.  

Does the NativeImport package load the C DLL when it is loaded?   If so, that might bump the ref count (or whatever actual implementation Microsoft used).  I think it is the CLR that is holding it.   The annotations on each native method (in [NativeMethods.cs](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag.NativeImport/NativeMethods.cs)) explictly call out the DLL:

```csharp
[DllImport(DLL_NAME, EntryPoint = nameof(plc_tag_check_lib_version), CallingConvention = CallingConvention.Cdecl, ExactSpelling = true)]
public static extern int plc_tag_check_lib_version(int req_major, int req_minor, int req_patch);
```

That part with the `DllImport` is the stanza that I think may be holding the library loaded.  I don't know if it is lazy loading or not, but definitely if you call one of the C DLL functions, the library will be loaded.   If there is some sort of ref count, then that will bump it to at least one and keep it in memory.

timyhac, 2022-12-09:
The interop library ([Platform Invoke](https://learn.microsoft.com/en-us/dotnet/standard/native-interop/pinvoke)) does the dll loading for us.

Doing some googling, since they're using Windows, they could use either the [FreeLibrary](
https://stackoverflow.com/questions/2445536/unload-a-dll-loaded-using-dllimport) function or the [NativeLibrary class](https://learn.microsoft.com/en-us/dotnet/api/system.runtime.interopservices.nativelibrary.free?view=net-7.0) depending on which platform they're using.

It would be good to get some code example, as its not clear to me what the problem is.
At a high level, I get it, but I think this is a problem where the devil is in the detail.

For context, the .NET wrapper targets .NET Standard 2.0 - which is the most straightforward way to support widest number of platforms - x86, x64, arm arm64, .NET Framework / .NET Core, windows, linux, mac, android, etc but it does mean that a lot of the platform-specific APIs are not available.

jkoplo, 2022-12-15:
I think I referenced it before, but I think the issue (and potentially the solution) are captured here: [StackOverflow: How to dynamically load and unload (reload) a .dll assembly](https://stackoverflow.com/questions/63616618/how-to-dynamically-load-and-unload-reload-a-dll-assembly).

Specifically, that answer points to this library: [.NET Plugins](https://github.com/natemcmaster/DotNetCorePlugins).

solax76, 2022-12-15:
I have the same issue and in my case it is solved by disposing every Tag I use and then call the libplctag.NativeImport.plctag.plc_tag_shutdown()

kyle-github, 2022-12-20:
There is a discussion about a very similar problem, native library not unloading, in the JNA mailing list I am on.   

> - the user had a relative path, which prevented the cache from loading the existing Java object
> - the loading wasn't also using the classloader to match up the options
>
> This meant the native LoadLibraryExW() was called a second time, incrementing the counter.  So even though we got the correct handle to the module, the dispose() (or close() in 5.12) call which called FreeLibrary() only decremented it once.

Is there a straightforward way to find out all the places that LoadLibraryExW() and similar functions are called as well as the places that FreeLibrary() is called?  Maybe there is a simple way to just keep a counter to find out if the count is being left higher than zero?

The .Net library probably isn't having the same kinds of cache and class loader points as the above, but it sets some context.

timyhac, 2022-12-20:
Yikes, I haven't done anything like this before. This StackOverflow article has some relevant points: https://stackoverflow.com/questions/864839/monitoring-certain-system-calls-done-by-a-process-in-windows

Nick.ko...@runb3.com - is it possible for you to provide a code example, I think we are prematurely finding solutions without understanding the problem.

timyhac, 2023-01-25:
None of the .NET runtimes automatically unload assemblies that are unused, so I think it is out of scope for libplctag.NET to provide additional support for this use case.

As Jody said earlier, a plugin model is what is required here.

In addition to his great notes, here are some other resources on the topic:
* https://learn.microsoft.com/en-us/dotnet/core/tutorials/creating-app-with-plugin-support
* https://learn.microsoft.com/en-us/dotnet/api/system.runtime.loader.assemblyloadcontext?view=net-7.0
* https://learn.microsoft.com/en-us/dotnet/standard/assembly/load-unload
* https://learn.microsoft.com/en-us/sysinternals/downloads/listdlls
  e.g. `.\Listdlls64.exe MyApplicationThatIsRunning.exe`

Using the ListDlls application along with this test application shows that plctag.dll does indeed stay in memory after a call to `plc_tag_shutdown()`

```csharp
Console.WriteLine("Stage1: plctag.dll shouldn't be in memory, press any key to load it");
Console.ReadLine();

var major = libplctag.NativeImport.plctag.plc_tag_get_int_attribute(0, "version_major", 0);
var minor = libplctag.NativeImport.plctag.plc_tag_get_int_attribute(0, "version_minor", 0);
var patch = libplctag.NativeImport.plctag.plc_tag_get_int_attribute(0, "version_patch", 0);
var version = $"v{major}.{minor}.{patch}";
Console.WriteLine(version);

Console.WriteLine("Stage2: plctag.dll was loaded into memory by retrieving the verson number");
Console.WriteLine();

Console.WriteLine("Press any key to Shutdown libplctag");
Console.ReadLine();

libplctag.NativeImport.plctag.plc_tag_shutdown();

Console.WriteLine("Stage3: libplctag shutdown was called, is plctag.dll still in memory?");
Console.Read();
```

[OutputOfConsoleApp1.exe.txt](https://github.com/libplctag/libplctag.NET/files/10504353/OutputOfConsoleApp1.exe.txt)
[AfterStage1.txt](https://github.com/libplctag/libplctag.NET/files/10504352/AfterStage1.txt)
[AfterStage2.txt](https://github.com/libplctag/libplctag.NET/files/10504351/AfterStage2.txt)
[AfterStage3.txt](https://github.com/libplctag/libplctag.NET/files/10504350/AfterStage3.txt)

closed 2023-01-25 by timyhac