libplctag/libplctag.NET #318: Try Notification value changed

patopat, 2022-12-19:
Hi,

In my PlcClient.class :
```Csharp
        public void SubscribeNotificationChangedTag(EventHandler<EventArgs> eventHandler)
        {
            var myUdtTag = new NotifyValueChangedTag<RunStopRecordingPlcMapper, RunStopRecordingPlc>()
            {
                Name = "Recording",
                Gateway = Gateway,
                Path = Path,
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                AutoSyncReadInterval = TimeSpan.FromMilliseconds(1000)
            };

            myUdtTag.ValueChanged += eventHandler;
        }

```

In my Programs :
```Csharp
 MyInit()
{
    _plcClient.SubscribeNotificationChangedTag(RunStopRecordingEvent);
}
        private void RunStopEvent(object sender, EventArgs e)
        {
            if (sender is RunStopLoggerPlc)
            {
                RunStopLoggerPlc runStopLoggerPlc = (RunStopLoggerPlc)sender;
            }
}

```

timyhac, 2022-12-19:
Hi @patopat - this isn't enough information for us to help - could you please explain more clearly what problem you are having.

To begin with, can you please provide the following:

1. You said "I have warning in MyInit method" - what is the warning, and where is this warning coming from (i.e. is it a compiler warning, a runtime warning)?
2. Is there a problem with your code, or are you asking for code review suggestions?
3. If there is a problem; what are you expecting to happen, and how does your code behave differently to this expectation?
4. Provide a complete code example that shows this.

patopat, 2022-12-20:
@timyhac ,

Sorry, I was not very clear. 
I have already created the UDT tag mapper class.
Now, I would like subscribe on a 'bit' called Recording changed value and fire an event.
I would like pass the callback as parameter to the PLC client class.

First,  I suppressed the warning by adding the nullable object parameter as following :
```Csharp
        private async void RunStopRecordingEvent(object? sender, EventArgs e)
        {
            RunStopRecordingPlc runStopRecordingPlc = (RunStopRecordingPlc)sender;
            ....
        }
```
Secondly, is there the right cast to get the data from PLC ?

Regards,
Pat

timyhac, 2022-12-20:
I think you're asking what the Type of `sender` is on the `ValueChanged` event. So assuming you've more or less copied [this example](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ExampleNotifyChanged.cs) you'll note that `this` is a reference to the [`NotifyValueChangedTag`](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ExampleNotifyChanged.cs#L60).

This is code in your project that I cannot see, but it does mean you can set a breakpoint and inspect the type yourself.

patopat, 2022-12-20:
Hi @timyhac ,

Unfortunately, Now I can't test with the real hardware, perhaps next year.

I was wondering if the data from the PLC is passed through the object sender or the EventArgs parameter ? And if there are raw data and have to decode or just a cast is enough ?

Pat

closed 2022-12-29 by patopat