libplctag/libplctag.NET #320: Get ITag object from TagInfo c# .NET framework

scottdf93, 2023-01-04:
I can't work out from the examples how to get a tag from the TagInfo.

I'm using the below to get all the tags

`            // Create the tags class
            Tag<TagInfoPlcMapper, TagInfo[]> controllerTags = new Tag<TagInfoPlcMapper, TagInfo[]>()
            {
                Gateway = this.IpAddress.ToString(),
                Path = this.ControllerPath,
                PlcType = this.Type,
                Protocol = this.Protocol,
                Name = "@tags"
            };`

then reading in the `.Value` property to get the TagInfo objects. Now I want to convert each of them to a Tag object that I can read the value of (ITag, TagBool, TagDint etc)

timyhac, 2023-01-05:
Following [this example](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ListUdtDefinitions.cs), you will need to create a Tag object for every tag you want to read. The info returned in the TagInfos has what you need to create them.

What have you tried so far?

scottdf93, 2023-01-05:
At the moment i'm just trying to create a tag object then attempt to read within a try/catch, if it fails I then try a different object type but I'm finding now that lots of tags are now coming in as the wrong data types.

[image]

I want to get the ITag from from controllerTags based of the tag name

jkoplo, 2023-01-07:
I think you're probably getting two different things a little mixed together - are you coming from the Ingear AB library by chance?

With libplctag you'll want to read the `@tags` tag. It will give you information about the tags on the controller - not a 'prototype' for each tag. If I remember, you may need to recurse on some of the non-base tags to get a full tag map. I'm not on my development machine, but I might have an example that I can share (and add to our documentation).

timyhac, 2023-01-22:
Hi @scottdf93 - I trust this has answered your question, please raise a new one if you are still unable to create tags from the tag listing information and would like support from us.

closed 2023-01-22 by timyhac