libplctag/libplctag.NET #321: Writing to custom length string

airwedge1, 2023-01-09:
It seems reading a string longer then the default 82 is possible by default but writing to a string returns an error.  Reading over the documentation in the base libplctag project the str_max_capacity and str_total_length have to be set to do so, but they are not exposed in TagOfT.  Exposing them would make this process much easier.

Attached is a patch where I add the properties to TagOfT and and CustomStringExample.cs to how it would be used. 

Let me know your thoughts.
[custom-length-string.patch](https://github.com/libplctag/libplctag.NET/files/10375982/custom-length-string.patch)

kyle-github, 2023-01-09:
It depends what kind of PLC you have.  If it is a ControlLogix or a CompactLogix and you are using standard STRING tags, then setting these values will not allow you to write a longer string.  Similarly, if it is a DF1/PCCC PLC like PLC/5, SLC 500 or MicroLogix, then you won't be able to make the string longer.

If it is a Micro800 series, then I think the library defaults may be wrong and you could theoretically set the string size to 255 (maybe, I don't have hardware to test).

airwedge1, 2023-01-09:
> On a system that lets you define your own string-like UDTs, you can set the various string attributes (when libplctag.Net exposes them).

Thanks for your replies.  Correct the system I am using allows to define your own string-link UDT.  Setting the max and total attributes allow it to read and write beautifully.  My patch that I included simply exposes those attributes so they can be adjusted and used.

timyhac, 2023-01-09:
These properties are exposed on the non-generic [Tag](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/Tag.cs).

The purpose of Tag<M,T> is to combine the logic of a Mapper with the underlying Tag and present the tag value as a CLR object - if we exposed all of the other Tag configuration options you could set the string length on a DINT tag for example - doesn't make sense.

The right place to put this is with the mapper or with a subclassed type - there is a PR to allow mappers to configure these properties - but it hasn't been merged yet: https://github.com/libplctag/libplctag.NET/issues/211

airwedge1, 2023-01-09:
Ok.  Fair enough.

closed 2023-01-09 by airwedge1