libplctag/libplctag.NET #326: Custom UDT mapper with string

patopat, 2023-01-30:
Hi,
Could you provide a short example about how to write a char array of size 20 from my Custom UDT because the CompactLogix controller 5370 doesn't know what is a string ?

```Csharp
    public class CustomTagPlcPlcMapper : PlcMapperBase<CustomTagPlc>, IPlcMapper<CustomTagPlc>
    {
        public override int? ElementSize => 4 + 20;

        public override CustomTagPlc Decode(Tag tag, int offset)
        {
            var DINT0 = tag.GetInt32(offset + 0);
            var STRING0 = tag.GetString(offset + 4);

            var customTag = new CustomTagPlc
            {
                Id= DINT0,
                Datetime = STRING0
            };

            return customTag;
        }

        public override void Encode(Tag tag, int offset, CustomTagPlc value)
        {
            tag.SetInt32(offset + 0, value.Id);
            tag.SetString(offset + 4, value.Datetime);
        }
    }

    public class CustomTagPlc
    {
        public int Id { get; set; }
        public string Datetime { get; set; }
    }
```
Regards,
Pat

timyhac, 2023-01-31:
The code before change might be useful: https://github.com/libplctag/libplctag.NET/commit/fc7c18b374a60590896180346e62d1906cfd070b.

Be careful that you're using the right encodings - it looks like we previously assumed ASCII encoding but make sure its appropriate to your application.

When developing the UDT martialling logic it can be useful to use the GetBuffer() method to access the entire buffer, and inspect it using a hexdump tool. I normally use [hexed.it](https://hexed.it/).

If you're looking to use the [string configuration properties](https://github.com/libplctag/libplctag/wiki/API#handling-strings) it might be possible to set them in the Encode/Decode functions.

patopat, 2023-01-31:
@timyhac ,

Yes, very useful thank you.

closed 2023-02-01 by timyhac