libplctag/libplctag.NET #328: Issue timeout on first read

patopat, 2023-02-03:
Hi,
Please, I need your help because I have a lot of timeout on first connection and I don't know why ?
I run the Example program from library.
The log file is attached.

[image]

[details.log](https://github.com/libplctag/libplctag.NET/files/10579336/details.log)

Regards,
Pat

patopat, 2023-02-03:
I set 30 seconds for timeout and I get this following error : 
[image]

timyhac, 2023-02-04:
Hi again - can you clarify what you mean by:

> a lot of timeout on first connection

i.e. 
* Does it always work if you try it twice, three times, four times?
* Does it work against other PLCs and not this one?
* Does it work against other Tags and not this one?
* Does it sometimes work against this tag, but not always? How often does it work (e.g. 20 times out of 100 attempts)

In the future could you provide a full code listing for your minimal reproducible example, there could be other factors at play.

Thanks for the debug log.

patopat, 2023-02-04:
@timyhac I wrote an application 2 months ago and everything worked very well. Now, I have a lot of difficulties to connect to the PLC. After one day of debugging I noticed that is coming from Windows 11 22H2. My colleague also has problem to Ping the PLC with Win11 22H2.

I will mount a VHDX with Win10 22H1 and will try.

Pat.

timyhac, 2023-02-05:
It sounds like you've confirmed the issue is elsewhere in the system?

patopat, 2023-02-07:
@timyhac 
Yes I confirm that the issue comes from Win11.

Pat

closed 2023-02-07 by timyhac