libplctag/libplctag.NET #338: Timeout Micro850 (2080-LC50-24QBB)

fiddlydigit, 2023-04-14:
I always get timeout trying to return a tag. I have tried removing Path but instead of Timeout I get BadParam exception. Am I missing something? Thanks in advance.

```csharp
public static Tag Test()
    {

        var myTag = new Tag()
        {
            Name = "Programs:SomeProgram.SomeDint",
            Path = "1,0",
            Gateway = "192.168.100.199",
            PlcType = PlcType.Micro800,
            Protocol = Protocol.ab_eip,
        
            //A global timeout value that is used for Initialize/Read/Write methods
            Timeout = TimeSpan.FromSeconds(1)
        };
        myTag.Initialize();
        return myTag;
    }
```

[image]

msienkow, 2023-04-14:
@fiddlydigit 

I'm pretty sure that libplctag can't read program scoped tags from the Micro800 series. I know the same issue exists with the various python Ethernet/IP libraries. However you can read globally scoped tags from the Micro800. Also, when using with the Mico800, you have to comment out the Path parameter. 

`public static Tag Test()
    {

        var myTag = new Tag()
        {
            Name = "SomeDint", // Globally scoped DINT tag.
            // Path = "1,0",
            Gateway = "192.168.100.199",
            PlcType = PlcType.Micro800,
            Protocol = Protocol.ab_eip,
        
            //A global timeout value that is used for Initialize/Read/Write methods
            Timeout = TimeSpan.FromSeconds(1)
        };
        myTag.Initialize();
        return myTag;
    }`

fiddlydigit, 2023-04-14:
I have tried not including Path but I get BadParameter exception (I posted logs). So there is no way of getting variables or DI/DO?

msienkow, 2023-04-14:
> I have tried not including Path but I get BadParameter exception (I posted logs). So there is no way of getting variables or DI/DO?

You have to use globally scoped tags on the MicroLogix. 

[image]

fiddlydigit, 2023-04-17:
@msienkow That was it🤦...Thanks!

closed 2023-04-17 by fiddlydigit