libplctag/libplctag.NET #340: Connecting to a topic name

kaptindan, 2023-05-15:
Hello, I have a PLC I am trying to communicate with. I can read the tags from the main PLC but this PLC is not directly on the same network but has access to the main. I can connect with RSLinx using the topic name but can not connect using libplctag IP address.

kyle-github, 2023-05-15:
Based on that picture, I think you want to use CIP bridging.   The path will probably be something like "1,3,18,192.168.12.2,1,0".  I'll break that down.

1,3 - this routes the traffic into the PLC to the backplane ("1") and then to the slot with the other network card, "3".

18,192.268.12.2 - this tell the network card to use port A (18 is A and 19 is B), and then route traffic to IP address 192.168.12.2 where the second PLC is.

1,0 - this tell the second PLC to route traffic to the backplane, "1", and then to slot 0 where the CPU is.

kaptindan, 2023-05-16:
Thanks for the help! I am new to this and trying to learn! So How do I reference the 192.168.11.1 PLC to get to the backplane? I tried that and got libplctag.LibPlcTagException: 'ErrorBadParam'

kaptindan, 2023-05-16:
Hey Kyle! Thanks so much for the information! I am still a learner. This helped a lot.

closed 2023-05-20 by timyhac