libplctag/libplctag.NET #341: Running on .NET Alpine docker image

Lazer91, 2023-05-23:
I'm struggling to make this work inside an Alpine-based .NET docker image, due to the dependency on `glibc`. Here's what I tried:

**This works** (it's Debian)
```
FROM mcr.microsoft.com/dotnet/aspnet:7.0
COPY bin/Release/net7.0/publish ./app
ENTRYPOINT ["dotnet", "/app/MyApp.dll"]
```

**This doesn't work**, `glibc` is missing in Alpine
```
FROM mcr.microsoft.com/dotnet/aspnet:7.0-alpine
COPY bin/Release/net7.0/publish ./app
ENTRYPOINT ["dotnet", "/app/MyApp.dll"]
```

I tried to add `gcompact` as per [THIS](https://wiki.alpinelinux.org/wiki/Running_glibc_programs) but **doesn't work** (segmentation fault as soon as the lib is invoked in MyApp)
```
FROM mcr.microsoft.com/dotnet/aspnet:7.0-alpine
RUN apk add --no-cache gcompat libstdc++
COPY bin/Release/net7.0/publish ./app
ENTRYPOINT ["dotnet", "/app/MyApp.dll"]
```
I tried also [sgerrand/alpine-pkg-glibc](https://github.com/sgerrand/alpine-pkg-glibc):
```
FROM mcr.microsoft.com/dotnet/aspnet:7.0-alpine
RUN wget -q -O /etc/apk/keys/sgerrand.rsa.pub https://alpine-pkgs.sgerrand.com/sgerrand.rsa.pub && \
   wget https://github.com/sgerrand/alpine-pkg-glibc/releases/download/2.35-r1/glibc-2.35-r1.apk  && \
   apk add glibc-2.35-r1.apk
COPY bin/Release/net7.0/publish ./app
ENTRYPOINT ["dotnet", "/app/MyApp.dll"]
```
but it **doesn't work**, I get [THIS](https://github.com/sgerrand/alpine-pkg-glibc/issues/194) issue.

Has anyone managed to make this work in an Alpine image? I read that it's not recommended to use `glibc` in Alpine due to the presence of `musl`.

timyhac, 2023-05-23:
Do the libplctag core binaries work by themself for this setup?

kyle-github, 2023-05-23:
They might not.  You'll probably need to build them using the musl libraries.

Lazer91, 2023-05-25:
Indeed we solved by building the native library in an Alpine VM. Here are the commands if anyone is interested:
```
apk update
apk add build-base gcc git cmake

git clone https://github.com/libplctag/libplctag.git

cd libplctag
mkdir -p build
cd build

cmake .. -DCMAKE_BUILD_TYPE=Release
make
```
Please note that:
- The output `libplctag.so` that you will find in the output directory is just a 0 byte symbolic link. You will need to find the file it is pointing to (for example `libplctag.so.2.5`) and rename it to `libplctag.so`, then include it in your .NET solution with the "Copy always" action set.
- To avoid the unwanted overwriting of the file at the program startup, the following line of code is required in your app (before any use of the libplctag library), as explained [HERE](https://github.com/libplctag/libplctag.NET/blob/master/docs/Using-a-non-packaged-version-of-the-native-libplctag-library.md)

```
plctag.ForceExtractLibrary = false;
```

closed 2023-05-25 by Lazer91