libplctag/libplctag.NET #342: ReadAsync + Dispose

VitorKawao, 2023-06-01:
I am reading many tags async (with tag.ReadAsync()), it works great but if I dispose (tag.Dispose()) them I get this exception: "The process was terminated due to an unhandled exception." at libplctag.NativeTagWrapper

Probably because it disposed the tag before the read finished

Application: RunTime.exe
CoreCLR Version: 6.0.1523.11507
.NET Version: 6.0.15
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AggregateException: One or more errors occurred. (Cannot access a disposed object.
Object name: 'libplctag.NativeTagWrapper'.)
 ---> System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'libplctag.NativeTagWrapper'.
   at libplctag.NativeTagWrapper.iLimvgLBP3b()
   at libplctag.NativeTagWrapper.Abort()
   at libplctag.NativeTagWrapper.<>c__DisplayClass125_0.mKgm5GuiCMe()
   at System.Threading.CancellationToken.<>c.<Register>b__12_0(Object obj)
   at System.Threading.CancellationTokenSource.Invoke(Delegate d, Object state, CancellationTokenSource source)
   at System.Threading.CancellationTokenSource.CallbackNode.<>c.<ExecuteCallback>b__9_0(Object s)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.CancellationTokenSource.CallbackNode.ExecuteCallback()
   at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
   --- End of inner exception stack trace ---
   at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
   at System.Threading.CancellationTokenSource.NotifyCancellation(Boolean throwOnFirstException)
   at System.Threading.CancellationTokenSource.TimerCallback(Object state)
   at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
   at System.Threading.TimerQueue.FireNextTimers()

timyhac, 2023-06-01:
Hi Vitor - good to hear from you again!

Are you able to provide a code sample that reproduces this?

VitorKawao, 2023-06-02:
Hello Tim, 

I have this issue with my product, I could not have time yet to create a small project to reproduce it.
I'll try to explain our product.
It has lots of modules. One module is driver communication.
So this module has a thread that reads lots of tags per time (the client chooses the time frequency).
The client can stop or restart this module, so we dispose the tags.
If the client restarts it, the exception is thrown

timyhac, 2023-06-03:
It is the intended behaviour to throw an error after a tag is disposed.
https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/NativeTagWrapper.cs#L380

If you do really want to dipose the tags between stop and restart, you will need to create new tags on restart.

VitorKawao, 2023-06-16:
Hello Tim,
I could fix my problem, I changed my code, so this exception is not crashing my app anymore

Thank you,
Vítor Guedes

closed 2023-06-16 by VitorKawao