libplctag/libplctag.NET #344: Why can't I read out the PLC data type after a "Read" although the data type is supplied with the read?

joed74, 2023-06-29:
Why can't I read out the PLC data type after a "Read" although the data type is supplied with the read?

```        
var btag = new Tag
     {
                Gateway = gateway,
                Path = "1,0",
                PlcType = PlcType.ControlLogix,
                Protocol = Protocol.ab_eip,
                Timeout = new TimeSpan(0, 0, 0, 0, 250),
                Name = "HMI_Administration.PData.ToHmi.EnableLoadNext"
      };
      btag.Read();
      btag.Type ???
```

According to the wireshark answer the datatype is supplied with the value:

[image]

c1 is the datatype, bool

timyhac, 2023-06-30:
İf that ever becomes possible in core libplctag, it will be exposed to the .NET wrapper - see this ticket for discussion
https://github.com/libplctag/libplctag/issues/252

timyhac, 2023-06-30:
@joed74  - please chime in on that other ticket with your use case. I will close this ticket out as its primarily a core libplctag concern.

closed 2023-06-30 by timyhac