libplctag/libplctag.NET #349: PowerFlex 753 communication over Ethernet. Is it possible?

gsp-dce, 2023-08-25:
I'm using libplctag.NET with some AB CompactLogix and it is working perfectly!

But what about AB Drives PowerFlex 753 and PowerFlex 525 over Ethernet?

Is it possible to use this same lib (maybe with some minor changes)?

raentzmi, 2023-08-25:
I would doubt it.  These drives do not expose a tag model in the same way a
AB CompactLogix does.  What may work (will leave that up to the community)
is what is called CIA (Class Instance Attribute).
Even if this CIA would work you will only get slower data not high speed
current etc as I suspect you're looking for.

It would be a very valuable thing to have in a library like this - and
would be valuable for all of CIP devices, not just drives.

timyhac, 2023-08-26:
libplctag does allow for "raw" tags. [This Pull Request](https://github.com/libplctag/libplctag.NET/pull/212/files#diff-577efc91c1d7667debebeaba0ea20e783d6dfa42140e2cc790b03d45450172c3) contains an example that shows what it would look like to make use of that feature. I'm not sure if that is enough to interface with the drives.

There have been some similar questions asked in the past:
* https://github.com/search?q=org%3Alibplctag+powerflex&type=issues
* https://groups.google.com/g/libplctag/search?q=powerflex

raentzmi, 2023-08-26:
I looked at the similar questions and nothing pops out as a solution.

Rockwell has released a new firmware for the 755T (v12) that has some
symbols and templates that look like controller UDT's.  What this will do
is allow data collection directly from the drive.  I don't know if these
will be presented to CIP as a UDT - they do show up in LiveData which makes
me think that maybe this will work.

Now for the downside - this will be probably implemented on newer firmware
and new drives so older drives may never get the new firmware.  For these
older drives there is a CIP proxy that describes the CIA addresses for
relevant items in the drive.

This is why what is really needed is a CIA EtherNet/IP component - then you
could get to any parameter in the drive - once again slower than CIP
direct but still gets data.

Maybe next year I can look deeper into this -right now I don't have the
time...

Where I would start is reading the CIP Identity object - this contains
pretty much static data about the device ...

timyhac, 2023-10-29:
Hi @gsp-dce  - thanks for the question, I trust you found an answer to your question!

@raentzmi - thanks for the extra information!

closed 2023-10-29 by timyhac